realsenseai/librealsense #4344: Length of bag file

Rizhiy, 2019-07-01:
Camera Model: D435
Language: Python

### Issue Description

How can I get the length of a `.bag` segment?
SDK doesn't have an obvious function and by default the video just loops back to the start.

realsens01, 2019-07-01:
Is [get_duration](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.playback.html#pyrealsense2.playback.get_duration) what you're looking for? If you want the number of frames, I believe in the past I used this in combination with ``seek`` and ``get_position``.

```
profile  = pipeline.start(config)
playback = profile.get_device().as_playback()
playback.pause()
playback.set_playback_speed(3)#5)
playback.seek(playback.get_duration())
N        = playback.get_position()
playback.seek(datetime.timedelta(0))
```

Currently, I use the ``rosbag`` package to process Realsense bag files.

Rizhiy, 2019-07-01:
I need number of frames

realsens01, 2019-07-01:
I apologize. The code I provided is old, and I had forgotten that ``get_position`` returns a time, instead of a frame number. This might be one of the reasons that I switched to the ``rosbag`` packages, which I've found to be much easier to use.

```

realsens01, 2019-07-01:
This worked for me, but I did not verify the number:

```
import pyrealsense2 as rs2
pipeline = rs2.pipeline()
config = rs2.config()
stream = rs2.stream.infrared
config.disable_all_streams()
config.enable_stream(stream)
config.enable_device_from_file('data.bag', repeat_playback=False)
profile = pipeline.start(config)
playback = profile.get_device().as_playback()
playback.pause()
playback.seek(playback.get_duration())
frame = rs2.composite_frame(rs2.frame())
if pipeline.poll_for_frames(frame): print(f'There are {frame.frame_number} {stream} frames in this bag.')
```

Rizhiy, 2019-07-02:
I just needed it to stop looping, `repeat_playback=False` worked for that.

closed 2019-07-02 by Rizhiy