realsenseai/librealsense #4346: non-exist points appear between two planes

Cavendish-Koo, 2019-07-02:
| Camera Model                       | D435 | 
| Operating System & Version |  Ubuntu 16.04  |                                      
| Platform                                 | PC |

### Issue Description
   I want to find edges of an object in front of the camera.However, when the  object is far away from the camera, the edges of the object are connected to the background with a list of points that actually not exist.It makes me hard to locate the object without depth mutation.
  The problem improved a little when I choose the high accuracy mode. But when the object is about 2m far from the camera, the situation appears again.
  I guess if the problem is caused by the error of stereo matching.I hope someone can help me.

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

@Cavendish-Koo  What's the SDK version and camera firmware version? Which tool/sample do you use when see the issue? And could you please provide some snapshot of the results so that we can understand the issue more clearly? Thank!

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

@Cavendish-Koo Could you please update? Thanks!

Cavendish-Koo, 2019-07-26:
[image]
Above is the unexpected image.The object connect with the wall behind it.

[image]
This is the good image, even though some of the edge connect with the wall too. It is about 2 meters far away from the object.

ev-mp, 2019-07-28:
@Cavendish-Koo , you can discriminate whether the artifacts are in the "raw" depth data or produced by post-processing by:
- Check and disable/adjust depth post-processing filters (if any). Turn off hole-filling filter/options if enabled.
- Navigate to the viewer's 3D upper right corner -  there is a button that selects pointcloud visualization as points or polygons (three mounted cubes) - switch it to point representation.

In case you still observe significant artifacts in the "naked" depth data, then you can share the rosbag for review. In that case please post the metrics obtained with the [Depth-Quality-Tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality).

ev-mp, 2019-09-19:
@Cavendish-Koo, I'm closing this due to lack of activity.
Feel free to reopen if you need additional assistance.

closed 2019-09-19 by ev-mp