realsenseai/librealsense #4349: C# How to get Colorized DepthFrame as RGBA

antongit, 2019-07-02:
Hello,

it's just a question. 
As far as I get it right if I apply Colorizer ProcessingBlock the resulting Frame has RGB Format.
Is it somehow possible to get this Frame as RGBA or convert the resulting Frame from RGB to RGBA afterwards? 

I think I'm missing something.

Thank you,
Anton

dorodnic, 2019-07-22:
There is currently no such option, but we will add it in one of the future releases

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

We plan to implement this. But there is no fix schedule yet. Thanks for your understanding.

MartyG-RealSense, 2022-04-23:
Hi @antongit  I would like to update you about the Intel feature request relating to this case.  It remains active but there is no progress to report at the time of writing this.

MartyG-RealSense, 2026-10-25:
Case closed due to the significant period of time that has passed.

closed 2026-10-25 by MartyG-RealSense