realsenseai/librealsense #4350: UDEV rules issue after updating to 2.24

kevindehecker, 2019-07-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   Ubuntu 18.04.01  | 
| Kernel Version (Linux Only)    |4.15.0-54-generic                              | 
| Platform                                 | PC  |
| SDK Version                            |  2.24                         | 

### Issue Description
After updating to 2.24 I started the realsense-viewer, which gives me the following warning:
[image]

```
Timestamp: 1562104878089.143555
Severity: Warn
Description: RealSense UDEV-Rules file:
 /lib/udev/rules.d/60-librealsense2-udev-rules.rules
 is not up-to date!
UDEV-Rules permissions configuration 
 for RealSense devices.

Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors
unless the application is running under 'sudo' (not recommended)
In case of Debians use: 
sudo apt-get upgrade/install librealsense2-udev-rules
To manually install UDEV-Rules in terminal run:
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger
```
It is also not detecting the camera that is connected and was detected just fine before all this. I've diffed the udev rules file with the current one in the repo, no differences except the name (which starts with 60 instead of 99)

The problem persists over rebooting. I can reproduce on both my desktop and my laptop. Completely sudo apt remove/install librealsense* does not help and does not show errors. Log:

```
~$ sudo apt remove librealsense2* 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'librealsense2-dbg' for glob 'librealsense2*'
Note, selecting 'librealsense2-dev' for glob 'librealsense2*'
Note, selecting 'librealsense2' for glob 'librealsense2*'
Note, selecting 'librealsense2-gl-dbg' for glob 'librealsense2*'
Note, selecting 'librealsense2-gl-dev' for glob 'librealsense2*'
Note, selecting 'librealsense2-gl' for glob 'librealsense2*'
Note, selecting 'librealsense2-utils' for glob 'librealsense2*'
Note, selecting 'librealsense2-dkms' for glob 'librealsense2*'
Note, selecting 'librealsense2-udev-rules' for glob 'librealsense2*'
Package 'librealsense2-gl-dbg' is not installed, so not removed
Package 'librealsense2-gl-dev' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libatk-bridge2.0-dev libatspi2.0-dev libdbus-1-dev libepoxy-dev libglfw3
  libgtk-3-dev libxkbcommon-dev libxtst-dev wayland-protocols
  x11proto-record-dev
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  librealsense2 librealsense2-dbg librealsense2-dev librealsense2-dkms
  librealsense2-gl librealsense2-udev-rules librealsense2-utils
0 upgraded, 0 newly installed, 7 to remove and 5 not upgraded.
After this operation, 200 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 304695 files and directories currently installed.)
Removing librealsense2-dbg:amd64 (2.24.0-0~realsense0.1244) ...
Removing librealsense2-utils:amd64 (2.24.0-0~realsense0.1244) ...
Removing librealsense2-dev:amd64 (2.24.0-0~realsense0.1244) ...
Removing librealsense2-dkms (1.3.5-0ubuntu1) ...

-------- Uninstall Beginning --------
Module:  librealsense2-dkms
[... 176 lines cut ...]
   - Installing to /lib/modules/4.15.0-54-generic/updates/dkms/

depmod...

DKMS: install completed.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.5, 4.15.0-54-generic, x86_64: installed
nvidia, 390.116, 4.15.0-52-generic, x86_64: installed
nvidia, 390.116, 4.15.0-54-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.5
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
Setting up librealsense2-dev:amd64 (2.24.0-0~realsense0.1244) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Setting up librealsense2-dbg:amd64 (2.24.0-0~realsense0.1244) ...
Setting up librealsense2-gl:amd64 (2.24.0-0~realsense0.1244) ...
Setting up librealsense2-gl-dev:amd64 (2.24.0-0~realsense0.1244) ...
Setting up librealsense2-gl-dbg:amd64 (2.24.0-0~realsense0.1244) ...
Setting up librealsense2-utils:amd64 (2.24.0-0~realsense0.1244) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...

```

ijsmallman, 2019-07-03:
I have this issue too

ev-mp, 2019-07-04:
@kevindehecker  thanks for report.
Seem be due inconsistency in release vs debug builds. We're working on it.
As for now - the specific pop-up is a false alarm, it doesn't not affect functionality and can be safely disregarded

beduffy, 2019-07-04:
I have the same issue, should I revert back to 2.23?

bsamadi, 2019-07-05:
realsense-viewer cannot see 435i or T265. It affects functionality.

zinan93, 2019-07-08:
I also have the same issue, cannot see d435

kevindehecker, 2019-07-08:
#sudo apt-cache policy  librealsense2 #pick v2.23
sudo apt remove librealsense2*
sudo apt-get install librealsense2=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-dev=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-dbg=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-gl*=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-udev*=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-utils=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-dkms

BTW, for me the warning in the realsense-viewer still does appear, but I can actually use both D435 and D435i's like normal.

atyshka, 2019-07-08:
Yeah having this issue too. However it seems that the rules are actually not installed. Running `ls /etc/udev/rules.d/` does not show anything related to realsense, and yet the device seems to work fine except for the viewer warning

kevindehecker, 2019-07-08:
They are here: https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules
If you copy that file into /etc/udev/rules.d/ and reload the udev rules: `udevadm control --reload-rules && udevadm trigger` maybe it will work again...

gtiongco95, 2019-07-08:
Same as other people. No rules installed. Warning appears. Camera not detected/working.

ev-mp, 2019-07-09:
@gtiongco95 hello,
The udev rules files will be installed into the ``/etc/udev/rules.d" and/or "/lib/udev/rules.d" based on the type of installation selected (manual or via debian packages accordingly).
The purpose of this notification is to alert the user in case one has outdated or even multiple udev files in those locations which can lead to unexpected behavior.

Make sure to follow pickup the udev step from the installation guide. For debian distro - running "dpkg -L librealsense2-udev-rules" will show the full location of the installed file.
Regarding the false warning - the fix PR has already been merged into the development branch, so in case you've resolved the udev installation issue, rebuilding the SDK from source should also remove the notification.

tarekmuallim, 2019-07-22:
I have the same problem.
Also Realsense viewer can't detect any d400 camera.

nyanko-box, 2019-07-24:
Hello, there.

I have the same issue under following environment (Ubuntu 16.04 LTS) .  

```
librealsense2/xenial,now 2.24.0-0~realsense0.1243 amd64 [インストール済み、自動]
librealsense2-dbg/xenial,now 2.24.0-0~realsense0.1243 amd64 [インストール済み]
librealsense2-dev/xenial,now 2.24.0-0~realsense0.1243 amd64 [インストール済み]
librealsense2-dkms/xenial,now 1.3.5-0ubuntu1 all [インストール済み]
librealsense2-gl/xenial,now 2.24.0-0~realsense0.1243 amd64 [インストール済み、自動]
librealsense2-gl-dbg/xenial 2.24.0-0~realsense0.1243 amd64
librealsense2-gl-dev/xenial 2.24.0-0~realsense0.1243 amd64
librealsense2-udev-rules/xenial,now 2.24.0-0~realsense0.1243 amd64 [インストール済み]
librealsense2-utils/xenial,now 2.24.0-0~realsense0.1243 amd64 [インストール済み]
```

My d435 is not found in realsense viewer v2.24.0, even via other USB port (2.0 and 3.0) or other Type-C2A USB cable.

I tried degrading above library version to 2.23.0-0~realsense0.1165, this issue still occured.

This d435 is found and works in other PC (Windows 10).

nyanko-box, 2019-07-24:
Hello, there.

Above comment of mine is solved by following procedure.

1. turning off "secure boot function" on BIOS
2. reinstall ubuntu
3. reinstall RealSense SDK

I wish this help for you!

afvillamil231, 2019-07-24:
Same problem here! I have already turned off the secure boot function before installing ubuntu and the SDK, and it does not work though. I cannot see the camera when using RTAB-map.

xudooo, 2019-07-25:
@nyanko-box 
turning off "secure boot function" on BIOS **WORKS** on my computer, thanks.

ev-mp, 2019-07-28:
@afvillamil231 , @tarekmuallim , @ijsmallman, @fgolemo , @gtiongco95 , any updates?
 In case you still experience issues with camera not being recognized/working please provide the output of `rs-enumerate-devices` and `dmesg -wT` when connecting the camera and run the tool.

Also specify the platform, sdk and kernel versions used to make you're not shifting the thread's focus.

tarekmuallim, 2019-07-28:
I reinstall the sdk and realsense viewer is detecting the cameras right now.
Thank you for your care.

SainaRez, 2019-07-30:
I ran the command:

sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger

and removed the duplicate:

sudo rm /lib/udev/rules.d/60-librealsense2-udev-rules.rules

the problem is solved and I don't get the warning anymore.

kevindehecker, 2019-08-04:
Just tested this with release of 2.25, seems to be fixed! So, closing this.

closed 2019-08-04 by kevindehecker

mathiassteger, 2022-12-31:
I had to add a `sudo`-call at the end too. So in my case I had to 

`sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && sudo udevadm trigger`

Thanks **alot** for your comment though, it saved my night!