realsenseai/librealsense #4351: Unable to link after update to 2.24

kevindehecker, 2019-07-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0.54)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.24                          | 
| Language                            |  C }                          | 

### Issue Description
After Ubuntu automatically updated librealsense to 2.24, (and rebooting!), I'm unable to link with librealsense.

```
/usr/bin/ld: cannot find -lrealsense2
collect2: error: ld returned 1 exit status
CMakeFiles/lalala.dir/build.make:503: recipe for target 'lalala' failed
make[2]: *** [lalala] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/lalala.dir/all' failed
make[1]: *** [CMakeFiles/lalala.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

Before the update to 2.24 all was fine. Even after updating I could compile, but I tried to reboot because of #4350, after which the problem appeared. Have reproduced this problem on two computers.

kevindehecker, 2019-07-03:
I've reverted the packages back to v2.23, this solves the problem for now:

```
#sudo apt-cache policy  librealsense2 #pick v2.23
sudo apt remove librealsense2*
sudo apt-get install librealsense2=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-dev=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-dbg=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-gl*=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-udev*=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-utils=2.23.0-0~realsense0.1166
sudo apt-get install librealsense2-dkms
```

ev-mp, 2019-07-04:
@kevindehecker hello,
There were a missing dependency in the Debian build that has been resolved. Please try the latest and update

kevindehecker, 2019-07-04:
Yes seems to work again, thanks!

kevindehecker, 2019-07-04:
Crap, it is *not* working!

I tumbled into the reboot trap... After a reboot (needed for something else) I could *not* link once more.
BTW, the reported version after updating  is still 2.24.

ev-mp, 2019-07-04:
That's correct - the change is in the Debian build (number) - there is no change in Librealsense.
The only thing that changes after reboot is the the `source ...` and the linker list are being updated.
Is it possible that you have several versions installed in separate locations ?
Can you share the error log ?

kevindehecker, 2019-07-04:
Error log? You mean the linker error?

```
/usr/bin/ld: cannot find -lrealsense2
collect2: error: ld returned 1 exit status
CMakeFiles/lalala.dir/build.make:503: recipe for target 'lalala' failed
make[2]: *** [lalala] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/lalala.dir/all' failed
make[1]: *** [CMakeFiles/lalala.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Really not that helpfull, but I don't have other errors... Please specify.

kevindehecker, 2019-07-04:
Aha, found the problem. In my CMakeList.txt I set the realsense library as follows:

```
set(REALSENSE realsense2) 
target_link_libraries(${PROJECT_NAME} ${REALSENSE})
```

Probably got that from some example in the early days. However, if I properly find the realsense package, the linker problem is solved!

```
find_package(realsense2 REQUIRED)
target_link_libraries(${PROJECT_NAME} ${realsense2_LIBRARY})
```

closed 2019-07-04 by kevindehecker

danneboom, 2019-07-09:
Same problem here after 2.24 (2.24.0-0~realsense0.1243) update. Before, in **QT** I could link with

     LIBS += -lrealsense2

Now, it says 

    /usr/bin/ld: cannot find -lrealsense2

and also `LIBS += -L/usr/lib/x86_64-linux-gnu -lrealsense2` or `LIBS+=/usr/lib/x86_64-linux-gnu/librealsense2.so` does NOT work. Using pkg-config like 

    CONFIG+=link_pkgconfig
    PKGCONFIG+=realsense2

does not change anything. In **CMake** though, before 2.24 I was doing

```cmake
    find_library( REALSENSE_LIBRARY NAMES realsense2 PATHS /usr/local/lib)
    target_link_libraries(myLib ${REALSENSE_LIBRARY} )
```

That does not work anymore, even with new path `/usr/lib//usr/lib/x86_64-linux-gnu`! I have to use find_package now.

```cmake
    find_package( realsense2 REQUIRED)
    include_directories(  ${realsense2_INCLUDE_DIR} )
    target_link_libraries(myLib ${realsense2_LIBRARY} )
```

But in QT it still fails.

qihao96, 2019-07-22:
> I've reverted the packages back to v2.23, this solves the problem for now:
>
> ```
> #sudo apt-cache policy  librealsense2 #pick v2.23
> sudo apt remove librealsense2*
> sudo apt-get install [...]

When I try to apt-cache show <package-name>, realsense viersion I find is `2.23.0-0~realsense0.1166=5`

Can you test it ?
Thanks.

qihao96, 2019-07-22:
> When I try to `apt-cache show <package-name>`, realsense viersion I find is `2.23.0-0~realsense0.1166=5`
>
> Can you test it ?
> Thanks.

add ${realsense2_LIBRARY} to target_link_libraries() solve this bug.