realsenseai/librealsense #4352: Invalid postprocessing?

nevynwong, 2019-07-03:
| Camera Model                       | D415| 
| Firmware Version                   |11.6 | 
| Operating System & Version | Win10| 
| Platform                                 | PC|
| SDK Version                            |2.21| 
| Language                            |Matlab | 

I set post-processing in the viewer, but there is no post-processing for the depth and point cloud of re-convert.exe. Why?

ev-mp, 2019-07-04:
@nevynwong hi , the post-processing is an on-demand feature that is integrated into the customer code and is not a part of a core data streaming..
While the purpose of `rs-convert` tool is to transcode depth data into additional formats. Modifying the data on-the-fly is an extension use-case that is not in the scope of the tool.

You can privately modify `rs-convert` to also include depth post-processing, but it can require significant refactoring as the [Temporal filter](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#temporal-filter) need multiple (8) consecutive frames to run properly and the `rs-convert` tool tries to run as much frames as possible in parallel - therefore assumes that there is no inter-frame dependencies.

nevynwong, 2019-07-08:
@ev-mp hi ,It may be difficult for me to modify rs-convert.exe for post-processing, so how do I get the post-processing results?Is there any other way?
Thank you.

dorodnic, 2019-07-22:
You can see how each of the SDK tools are implemented and modify it to your needs. 
I'd recommend looking into python APIs.

closed 2019-07-22 by dorodnic