realsenseai/librealsense #4353: Multi Camera issue on Android 8.1.0

David-168, 2019-07-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Android  8.1.0| 
| Kernel Version (Linux Only)    |  Android 8.1.0                                         | 
| Platform                                 | Red Hydrogen One/Snapdragon 835  / etc..  |
| SDK Version                            |  {220 }                          | 
| Language                            |  Java+JNI                          | 
| Segment			|  {Smartphone/AR }                          | 

When I create a device like this and try to get the serial number from it. 
The  function always return the same one.  
MainActivity.java
.......
            for( int i=0; i< mRsContext.queryDevices().getDeviceCount();i++) {

                mPipeline[i] = new Pipeline();
                mConfig  = new Config();

                device=devices.createDevice(i);
                String serialnumber = device.getInfo(CameraInfo.SERIAL_NUMBER);
               mConfig.enable_device(serialnumber);
//               mConfig.enable_device(devicenmae[i]);
                mConfig.enableStream(StreamType.DEPTH, 640, 480);
                mConfig.enableStream(StreamType.COLOR, 640, 480);

                mPipeline[i].start(mConfig);
..................
config.c 
JNIEXPORT void JNICALL
Java_com_intel_realsense_librealsense_Config_nEnableDevice(JNIEnv *env, jclass type_, jlong handle,
                                                           jstring serialnumber) {
    rs2_error *e = NULL;
    const char *nativeString = (*env)->GetStringUTFChars(env, serialnumber, 0);
    rs2_config_enable_device((rs2_config *) handle, nativeString, &e);
    (*env)->ReleaseStringUTFChars(env, serialnumber, nativeString);
    handle_error(env, e);
}

I tried to use the serial number which is printed on bottom of the camera. The first camera works fine. But, the second camera doesn't start.

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@David-168 The latest SDK v2.31 has multi camera example for Android. https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/multicam
Please try to see if same issue on this. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@David-168 Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport