realsenseai/librealsense #4354: How to include librealsense2/rs.hpp

YH53, 2019-07-03:
Required Info
| Camera Model | { D415 } |
| Operating System & Version | {Win 10 |

| SDK Version | 2.0 |
| Language | C++ |
Platform: visual studio 2015

 I want to create a new project，but it appears:Fatal error C1083: Cannot open include file: "librealsense2/rs.hpp": No such file or directory.
I try to solve it by include the header files path into my Visual Studio.
the path of my sdk is:C:\Program Files (x86)\Intel RealSense SDK 2.0\include
I add them to Visual Studio C++ Include directories under Tools → Options → Projects and Solutions → VC++ Directories.
but it is still wrong，How to solve it ？  Can you tell me the correct process to add it.  Thanks a lot.

ev-mp, 2019-07-04:
@YH53 hello, I  could be wrong here but when added to `Tools → Options → Projects and Solution..` the compiler wil search only for headers that are declared with <name.h> and not "name.h".
I think adding the headers to the solution and not globally should suffice, either by adding this path the relevant vcproj-s or via a custom .props file.

YH53, 2019-07-05:
@ev-mp I‘m sorry I didn't understand what you meant，could you please describe it in detail. Thank you very much.

maipbui, 2019-07-08:
Copy all header files you need to 1 folder has same path with your solution. e.g. header files are in "include" folder and my project is "measure".
[image]
Then go to Project &rightarrow; Properties &rightarrow; C/C++ &rightarrow; Additional Include Directories &rightarrow; type in $(SolutionDir)

closed 2019-07-22 by dorodnic

votecoffee, 2020-10-22:
There are some ready to use project default property pages you can import for both RealSense and OpenCV at the following location:
C:\Program Files (x86)\Intel RealSense SDK 2.0
Open the property manager for your project (View > Property Manager or View > Other Windows > Property Manager)
Add intel.realsense.props and opencv.props to each build configuration.
This will add the misc bin/include/lib/linker settings you need for most things.