realsenseai/librealsense #4356: Get inclination angle of the camera

antoniodourado, 2019-07-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Linux Mint 19.1  | 
| Kernel Version (Linux Only)    |  4.15.0-54                                         | 
| Platform                                 | PC / NVIDIA Jetson  |
| SDK Version                            |  2.23.0                          | 
| Language                            |  C++                          | 
| Segment			|  Computer Vision                          | 

### Issue Description
Hi there.

I'm new to IMU stuff and my question is pretty simple. When I start gathering frames from realsense, is it possible to get its inclination angle right away? For example, if the camera is point to a parallel direction of the ground, the angle would be 90 degrees.

AnnaRomanov, 2019-07-04:
Hi, did you look at [rs_motion](https://github.com/IntelRealSense/librealsense/tree/master/examples/motion) example? It shows how to calculate the rotation in euler angles, would that help?

ev-mp, 2019-07-09:
@antoniodourado hello,
Was the reference provided by @AnnaRomanov  helpful ?

antoniodourado, 2019-07-09:
I'm still testing it!

Will be back soon with results.

antoniodourado, 2019-07-12:
I managed to get IMU data on camera inclination/tilt after taking some minutes on [rs-motion](https://github.com/IntelRealSense/librealsense/tree/master/examples/motion) example as indicated.

Thank you so much for your help!

closed 2019-07-12 by antoniodourado