realsenseai/librealsense #4357: Issue was closed without resolution, again.

realsens01, 2019-07-03:
See [my unresolved issue](https://github.com/IntelRealSense/librealsense/issues/3878).

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

Hi @realsens01,

Let's clarify.
Are you able to run any python examples in librealsense?
Or, is below code not working for you?

    **import** pyrealsense2 **as** rs
    **import** numpy **as** np
    **import** cv2

    _# Configure depth and color streams_
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
    config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
    config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
    config.enable_record_to_file(**"test.bag"**)

    _# Start streaming_
    pipeline.start(config)

    **try**:
     WIN_TITLE1 = **'Color + Depth'**
     WIN_TITLE2 = **'Left IR + Right IR'**
     **while True**:

     _# Wait for a coherent pair of frames: depth and color_
     frames = pipeline.wait_for_frames()
     depth_frame = frames.get_depth_frame()
     color_frame = frames.get_color_frame()
     ir1_frame = frames.get_infrared_frame(1)
     ir2_frame = frames.get_infrared_frame(2)
     **if not** depth_frame **or not** color_frame:
     **continue**

     _# Convert images to numpy arrays_
     depth_image = np.asanyarray(depth_frame.get_data())
     color_image = np.asanyarray(color_frame.get_data())
     ir1_image = np.asanyarray(ir1_frame.get_data())
     ir2_image = np.asanyarray(ir2_frame.get_data())

     _# Apply colormap on depth image (image must be converted to 8-bit per pixel first)_
     depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

     _# Stack both images horizontally_
     images = np.hstack((color_image, depth_colormap))
     images2 = np.hstack((ir1_image, ir2_image))

     _# Show images_
     cv2.namedWindow(WIN_TITLE1, cv2.WINDOW_AUTOSIZE)
     cv2.imshow(WIN_TITLE1, images)
     cv2.namedWindow(WIN_TITLE2, cv2.WINDOW_AUTOSIZE)
     cv2.imshow(WIN_TITLE2, images2)

     key = cv2.waitKey(1)
     **if** key == ord(**'q'**) **or** key == 27: _# q or ESC_
     **break**

    **finally**:

     _# Stop streaming_
     pipeline.stop()

realsens01, 2019-07-18:
Hi @RealSenseCustomerSupport . 
Thanks for your message. I do not have access to the camera at the moment, but I will test this as soon as I do. However, I'm guessing that it will fail at ``pipeline.start``. If you compare your snippet to what I posted, you can see that they are pretty similar:

**The code you posted:**
```
import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
config.enable_record_to_file("test.bag")

# Start streaming
pipeline.start(config)
```

**The code I posted:**
```
import pyrealsense2 as rs2

device = rs2.context().devices[0]
serial_number = device.get_info(rs2.camera_info.serial_number)

pipeline = rs2.pipeline()
config = rs2.config()
config.enable_device(serial_number)
config.enable_stream(rs2.stream.infrared, 0)
config.enable_stream(rs2.stream.infrared, 1)
pipeline.start(config)
```

The primary difference between the snippets is the call to ``config.enable_device`` and the details of the ``config.enable_stream`` call. I've tried variations similar to what you posted, but I will confirm again.

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

Hi @realsense01,

Below code should be fine. Wait for your test update, you can also enable librealsense log to show more info by below command in the same terminal.
set LRS_LOG_LEVEL=DEBUG (windows) or export LRS_LOG_LEVEL=DEBUG (Linux)

realsens01, 2019-07-23:
@RealSenseCustomerSupport . If that code is fine, then the answer is: yes, the code is not working for me... as stated in my original post. I'll post the log output.

bmar45, 2019-07-25:
You used :
`config.enable_stream(rs2.stream.infrared, 0)`
`config.enable_stream(rs2.stream.infrared, 1)`

If i am not mistaken, you have to do 
`config.enable_stream(rs2.stream.infrared, 1)`
`config.enable_stream(rs2.stream.infrared, 2)`

I think i had the same trouble and found out that a stream id of 0 provoked an error in my code. Using 1 and 2 instead of 0 and 1 fixed it.

realsens01, 2019-07-25:
My original post says that I tried that, but I wrote that almost 3 months ago, so I don't remember doing it. It wouldn't hurt to try again. In any case, thanks for your suggestion @bmar45 .

Quote from original:
> A reply to another user question suggests using enable_stream with stream indices 1 and 2. This also didn't work. The only stream index that works is 1.

realsens01, 2019-08-29:
* Update to address some of the previous questions.
* This uses the stream indices suggested by @bmar45 
* This uses the code suggested by @RealSenseCustomerSupport 

```
In [3]: import pyrealsense2 as rs
   ...: import numpy as np
   ...:
   ...: # Configure depth and color streams
   ...: pipeline = rs.pipeline()
   ...: config = rs.config()
   ...: config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
   ...: config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
   ...: config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
   ...: config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
   ...: config.enable_record_to_file("test.bag")
   ...:
   ...: # Start streaming
   ...: pipeline.start(config)
   ...:
   ...:
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-3-f010363c7b0c> in <module>()
     12
     13 # Start streaming
---> 14 pipeline.start(config)

RuntimeError: Failed to resolve request. No device found that satisfies all requirements
```

realsens01, 2019-08-29:
Requested log output:
```
 08/08 18:18:10,248 DEBUG [15324] (context.cpp:99) Librealsense VERSION: 2.10.4
 08/08 18:18:10,264 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:10,264 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:10,264 DEBUG [15324] (ds5-factory.cpp:279) try_fetch_usb_device(...) failed.
 08/08 18:18:10,264 WARNING [15324] (win-uvc.cpp:967) Could not retrieve USB descriptor for device 8086:b07 , id:1e0da80f
 08/08 18:18:10,496 WARNING [15324] (win-uvc.cpp:967) Could not retrieve USB descriptor for device 8086:b07 , id:1e0da80f
 08/08 18:18:10,496 DEBUG [15324] (context.cpp:99) Librealsense VERSION: 2.10.4
 08/08 18:18:10,511 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:10,511 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:10,511 DEBUG [15324] (ds5-factory.cpp:279) try_fetch_usb_device(...) failed.
 08/08 18:18:10,511 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:10,527 DEBUG [15324] (ds5-factory.cpp:279) try_fetch_usb_device(...) failed.
 08/08 18:18:10,527 WARNING [15324] (win-uvc.cpp:967) Could not retrieve USB descriptor for device 8086:b07 , id:1e0da80f
 08/08 18:18:10,781 WARNING [15324] (win-uvc.cpp:967) Could not retrieve USB descriptor for device 8086:b07 , id:1e0da80f
 08/08 18:18:10,834 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,834 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,834 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,834 WARNING [15324] (sensor.cpp:315) Unregistered Media formats : [ UYVY Y12I Y8I  ]; Supported: [ ]
 08/08 18:18:10,881 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,881 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,881 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,881 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:10,897 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:10,897 DEBUG [15324] (ds5-factory.cpp:279) try_fetch_usb_device(...) failed.
 08/08 18:18:10,897 WARNING [15324] (win-uvc.cpp:967) Could not retrieve USB descriptor for device 8086:b07 , id:1e0da80f
 08/08 18:18:10,897 INFO [15324] (environment.cpp:85) Found fd unreachable streams, 4f1 extrinsics deleted
 08/08 18:18:11,166 WARNING [15324] (win-uvc.cpp:967) Could not retrieve USB descriptor for device 8086:b07 , id:1e0da80f
 08/08 18:18:11,229 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,229 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,229 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,229 WARNING [15324] (sensor.cpp:315) Unregistered Media formats : [ UYVY Y12I Y8I  ]; Supported: [ ]
 08/08 18:18:11,267 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,267 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,267 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,267 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,282 INFO [15324] (win-helpers.cpp:75) res=pSensorManager->GetSensorsByCategory(SENSOR_CATEGORY_ALL, &pSensorColl) returned: HResult 0x80070490: "Element not found."
 08/08 18:18:11,282 DEBUG [15324] (ds5-factory.cpp:279) try_fetch_usb_device(...) failed.
[... cut ...]
 08/08 18:18:11,637 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,637 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,637 INFO [15324] (win-helpers.cpp:75) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "The stream number provided was invalid."
 08/08 18:18:11,652 WARNING [15324] (rs.cpp:1426) null pointer passed for argument "profile"
```

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Hi @realsens01,

Your D435 camera behavior is quite weird. Per the log, the host even can't retrieve the USB descriptor. Suggest to upgrade your D435 firmware and librealsense and see how it works.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Hi @realsens01,

How is everything going?

realsens01, 2019-12-05:
I still can't record two IR streams.

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

Hi @realsense01,

I would suggest to have a RMA process for the failure analysis. You can check with your reseller. By the way, how many cameras do you have with this issue?

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @realsense01,

Haven't seen a response on this thread for a while.  Let us know your latest.

realsens01, 2020-01-13:
I purchased 4 D435 cameras directly from Intel. We use 2 at a time for data collection.

RealSenseCustomerSupport, 2020-01-13:
----------------------------------------------

Are you still seeing the original issue that was presented on the thread?

realsens01, 2020-02-06:
YES. I see the original issue on the four D435 cameras that I purchased directly from Intel.

RealSenseCustomerSupport, 2020-02-06:
----------------------------------------------

Hi @realsens01

Did you attempt to RMA the cameras with the return process on the site?

RealSenseSupport, 2020-08-20:
Hi @realsens01 

With the latest LRS and FW, does this issue continue?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-09 by RealSenseSupport