realsenseai/librealsense #4361: Template tracking

clear-zhu, 2019-07-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2019-07-04:
Is 'template tracking' a feature request, please?

clear-zhu, 2019-07-04:
I am using realsenseD435i to do a template tracking, which requires that I draw a rectangular box with the mouse on the area of interest in the video stream, save the area in the rectangular box as Mat format, and track the area. I know the methods and codes of ordinary cameras, and I am a novice in realsense. If you can provide a simple demo or example, I would like to ask you. Thank you very much

MartyG-RealSense, 2019-07-04:
Thank you for your additional information.  What you are aiming to achieve sounds complicated and so the project is probably best developed in stages instead of trying to do it all at once.

The first goal to achieve would be to draw the bounding box and exclude the data points outside of the box.  The link below describes the method that a RealSense user applied to achieve this.

https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157

clear-zhu, 2019-07-04:
It’s too complicated for me to say, I should say it step by step. Now I want to use the mouse to select the area of interest in the video stream. But I did not find the relevant documentation. (Because I want to identify a stationary object, this object is random, so I need to select the area with the mouse)

MartyG-RealSense, 2019-07-04:
Someone in the past asked about being able to select the ROI with the mouse in the RealSense Viewer.

https://github.com/IntelRealSense/librealsense/issues/2110

One of the Intel staff on this forum will be best equipped to give you advice on this subject.  I will tag a few of them into this conversation to highlight it to them.   Good luck!

@dorodnic @RealSenseCustomerSupport @RealSense-Customer-Engineering

eMrazSVK, 2019-07-04:
Hey @clear-zhu, as you're saying you're familiar with this problem, it should be sufficient enough for you, if you knew how to get frames from RealSense cameras. You can gather frames from camera with OpenCV like in this [example ](https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md). Then you have the color frame in Mat format and you can continue from there.

clear-zhu, 2019-07-08:
thanks your answer helped me a lot.

closed 2019-07-23 by RealSenseCustomerSupport