realsenseai/librealsense #4363: How to set or get ROI for Auto Exposure with C++

kazu0622, 2019-07-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  {2.20.0}                          | 
| Language                            |  C                          | 
| Segment			|  others }                         | 

Hello.
I'm trying to implement  custom ROI setting for Auto Exposure with C++. But found some sample code with python as mentioned in #4015, but not for C++.

I'd like to have an example of C++ description for designing custom ROI for Auto Exposure.

Thanks.

AntoineWefit, 2019-07-04:
Isn't [that](https://github.com/IntelRealSense/librealsense/blob/f112ed78e3917df79047254f864157195d4488b0/include/librealsense2/hpp/rs_sensor.hpp#L348-L379) C++ though ?
Hope this'll help !

kazu0622, 2019-07-04:
@AntoineWefit Thank you for your comment.
I described my code as below.
	rs2::region_of_interest roi;
	rs2::roi_sensor s(depth_sensor);
	roi = s.get_region_of_interest();
	s.set_region_of_interest(roi);
It worked!

Thank you.

closed 2019-07-04 by kazu0622