realsenseai/librealsense #4364: t265 relocalization feature unstable.

ahuizxc, 2019-07-04:
Recently I tested the export_localization_map and import_localization_map feature of t265, and there some problems about. Firstly, I set about 20 static node and export the localization map, then i imported the localization map and try to get all 20 nodes in a loop, once i got one node, then i would get all nodes, that's for sure, But here is the problem: even i set 20 nodes in the localization map, it is very hard to get even one static node in my environment, and when the localization map is to large, about 600,000, my  program will crash with LIBUSB_ERROR_TIMEOUT. Thanks~

RealSenseCustomerSupport, 2019-07-19:
----------------------------------------------

Hi @ahuizxc,

Can you share some code snippets in order to better understand your issue?

zainmehdi, 2019-12-03:
@RealSenseCustomerSupport After loop closing does it optimize the whole map or just the last pose. Can you please provide some insight. And it does optimize the whole map/trajectory is it possible to get it ?

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@harville , any comments about this?

RealSenseSupport, 2020-09-11:
Thank you for highlighting the issues with relocalization. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

zainmehdi, 2020-09-11:
I am really disappointed to hear that. It gives me an impression that other line of products will end up having the same fate.

RealSenseSupport, 2020-09-15:
The statement is targeted for the T265 product and not other products.

closed 2020-09-15 by RealSenseSupport