realsenseai/librealsense #4366: seg fault

schoningerJim, 2019-07-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Ubuntu 18.04 L4T  | 
| Kernel Version (Linux Only)    |    4.9.140-tegra   | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                        |  python                      | 
| Segment			|  Robot                           | 

### Issue Description
Basically, I have created a python camera class which create a camera by using the pyrealsense2 libraries
The constructor will init every stuff it has to be initialize ( pipeline, config,  and also create a thread which will grab the image using a while loop.
There is a start method which start the thread 
There is also a stop function which stop the thread and close the pipeline 

But anytime i call the close function, a seg fault happens!
Does someone have idea on why?
`

    def grab_img(cam):
         """This 'grab_img' function is designed to be run in the sub-thread.
        Once started, this thread continues to grab a new image and put it
         into the values, until 'thread_running' is set to False.
          """
          try:
               while cam.thread_running:
               frames = cam.pipeline.wait_for_frames()
               aligned_frames = cam.align.process(frames)
               aligned_depth_frame = aligned_frames.get_depth_frame()
               color_frame = aligned_frames.get_color_frame()
               cam.intrinsi = aligned_depth_frame.profile.as_video_stream_profile().intrinsics
              if not aligned_depth_frame or not color_frame:
			  continue
              cam.imgColor = np.asanyarray(color_frame.get_data())
              cam.imgDepth = np.asanyarray(aligned_depth_frame.get_data())
      
           finally:
               cam.thread_running = False
               cam.pipeline.stop()

   class Camera():

    def __init__(self):
		self.imgColor = None
		self.imgDepth = None
		self.thread = None
		self.thread_running = False
		self.pipeline = rs.pipeline()
		self.config = rs.config()
		self.config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
		self.config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
		self.profile = self.pipeline.start(self.config)
		self.depth_sensor = self.profile.get_device().first_depth_sensor()
		self.depthScale = self.depth_sensor.get_depth_scale()
		self.align_to = rs.stream.color
		self.align = rs.align(self.align_to)
		self.intrinsi = None

    def start(self):
        assert not self.thread_running
        self.thread_running = True
        self.thread = threading.Thread(target=grab_img, args=(self,))
        self.thread.start()

    def stop(self):
        self.thread_running = False
        self.thread.join()

    def read(self):
        return self.imgColor, self.imgDepth

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@schoningerJim Sorry for late response! What's SDK version when you get the issue? Did you try the latest SDK and FW to see if same issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@schoningerJim Did you still need help on this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@schoningerJim Sorry that we didn't get response from you for weeks. Will close the ticket at this point. Please feel free to create new ticket if you still have any questions/issues. Thanks!

closed 2019-10-25 by RealSenseCustomerSupport