realsenseai/librealsense #4368: T265 pose precision consultation

quxichao, 2019-07-05:
Hello, I want to ask about the positioning and direction error of the T265 camera.

good luck

ev-mp, 2019-07-09:
@quxichao , I'm not sure I understand what do you mean by that.
Can you be more specific and elaborate the actual use-case and the concern with the positioning/direction info ?

reyes73a, 2019-07-19:
Accuracy of any SLAM system is always in the order of multiple cm. I have measured T265 error to grow on average to 25cm after 25m motion and to 0.8m after 50-60m. Loop closure would help but it is not working that I know of.

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

We are always working on ways to improve and build upon the current accuracy of the T265.  Do monitor our releases which comes with target fixes or enhancements for stability and improvements on the algorithm.  There are many different environment features that can go into the performance of the T265 camera.

If there is a specific issue with details of user scenario available please open a new issue.  Thanks.

closed 2019-08-22 by RealSenseCustomerSupport