realsenseai/librealsense #4372: D435i can not work in MFC application

wenxyong, 2019-07-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Operating System & Version |   {Win 10  | 
                            | 
| Platform                                 | vs2015.  |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {VC ++}                          | 
               | 

### Issue Description
1.Open vs2015;
2,New project
3.Select MFC Application.
4.Create MFC Applcation.
5.Wirte folowing Code
void CMainFrame::OnM()
{
rs2::pipeline pipe;
pipe.start();
//rs2::config pipe_config;
// pipe_config.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
// pipe_config.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
// rs2::pipeline_profile profile = pipe.start(pipe_config);
while (1) {
	rs2::frameset data = pipe.wait_for_frames();//

	rs2::frame depth = data.get_depth_frame();//
	rs2::frame color = data.get_color_frame();

	const int depth_w = depth.as<rs2::video_frame>().get_width();
	const int depth_h = depth.as<rs2::video_frame>().get_height();
	const int color_w = color.as<rs2::video_frame>().get_width();
	const int color_h = color.as<rs2::video_frame>().get_height();
}
}
6.when run to
rs2::frameset data = pipe.wait_for_frames();//
throw error:
inline void error::handle(rs2_error* e)
{
if (e)
{
auto h = rs2_get_librealsense_exception_type(e);
switch (h) {
case RS2_EXCEPTION_TYPE_CAMERA_DISCONNECTED:
throw camera_disconnected_error(e);
case RS2_EXCEPTION_TYPE_BACKEND:
throw backend_error(e);
case RS2_EXCEPTION_TYPE_INVALID_VALUE:
throw invalid_value_error(e);
case RS2_EXCEPTION_TYPE_WRONG_API_CALL_SEQUENCE:
throw wrong_api_call_sequence_error(e);
case RS2_EXCEPTION_TYPE_NOT_IMPLEMENTED:
throw not_implemented_error(e);
case RS2_EXCEPTION_TYPE_DEVICE_IN_RECOVERY_MODE:
throw device_in_recovery_mode_error(e);
default:
throw error(e);
}
}
}
h value is RS2_EXCEPTION_TYPE_COUNT (8)
 
[image]

ev-mp, 2019-07-11:
@wenxyong , the application flow and also the attached log is not informative - what is the console output when the error occurs ?
One thing I noticed, though, is that you seem to start a new pipeline when event is fired.
In general, establishing a pipeline allocates all the internal resources and activates Host-Device streaming, therefore creating a pipeline should be performed only once on initialization. And during MFC's `::onEvent` routine you should call `pipe.wait_for_frames()`.

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

@wenxyong Could you please provide the console output when the error occurs as ev-mp requested? And also could you please follow ev-mp's suggestion about pipeline? Looking forward to your update. Thanks!

ev-mp, 2019-08-05:
Closed due to inactivity

closed 2019-08-05 by ev-mp