realsenseai/librealsense #4373: nodejs-pose.js failes, while the realsense-viewer wroks.

ijeppsen, 2019-07-07:
I've been trying to get the nodejs bindings to work with the t265 with no luck.  I've gotten and built from source the librealsense repo on Ubuntu 18.04.2 LTS and the realsense viewer loads and i'm able to see pose information from the device.  However after running 'npm install' in the wrappers/nodejs director without error, I'm unable to run any of the examples.  For instance:

` > node nodejs-pose.js ` 
returns:

/librealsense/wrappers/nodejs/examples/nodejs-pose.js:39
cfg = rs.config()
^

ReferenceError: rs is not defined
    at Object.<anonymous> (/home/ib/src/librealsense/wrappers/nodejs/examples/nodejs-pose.js:39:1)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:829:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
node: ../../libusb/io.c:2118: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
Aborted (core dumped)

Any suggestions would be appreciated.

RealSenseCustomerSupport, 2019-07-24:
----------------------------------------------

Hi Kevin,

I am not familiar with nodjs. Please help assign to others.

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

Hi @ijeppsen,

The implemented Node.js bindings of librealsense methods could be found in examples section:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs/examples

For full Reference Document please see:
https://github.com/IntelRealSense/librealsense/tree/development/wrappers/nodejs#3-api-reference-document

Please let us know it you need further clarifications regarding this topic.
Thank you!

closed 2019-11-11 by RealSenseCustomerSupport