realsenseai/librealsense #4375: Import Error SDK 2.24.0 Python3.7 on Raspberry Pi 4 Raspbian 10 Buster

baozinihao321, 2019-07-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.11.6 | 
| Operating System & Version |   Raspbian 10 Buster  | 
| Kernel Version (Linux Only)    |  4.19                                         | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  2.24.0                          | 
| Language                            |  python 3.7.3                    | 
| Segment			|  others                           | 

### Issue Description
I have already use cmake to build the source code and it finished 100%(although some warning during the period, such as -wformat-security)

cmake version 3.13.4

cmake command

cmake -DBUILD_EXAMPLES=false -DCMAKE_BUILD_TYPE=Release
           -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3) ..

IMPORT ERROR
>pi@raspberrypi:~ $ python3
>Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
>[GCC 8.2.0] on linux
>Type "help", "copyright", "credits" or "license" for more information.
> import pyrealsense2
>Traceback (most recent call last):
>File "<stdin>", line 1, in <module>
>ModuleNotFoundError: No module named 'pyrealsense2'

If your team don't support this platform yet, please let me know when you could update.
Thank you! All the best

CroMocha, 2019-07-07:
Have you tried steps 4 and 5 as found in:
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
Ran into this problem just recently too, easily solved by copying librealsense2.so that was in my librealsense/build directory and pyrealsense2.cpython-35m-arm-linux-gnueabihf.so (might be named differently in your installation) that was in librealsense/build/wrappers/python directory over to whichever directory you want to run your python scripts in.

baozinihao321, 2019-07-07:
Thanks for help. I update the .bashrc file.

export PYTHONPATH=$PYTHONPATH:/home/pi/librealsense/build/wrappers/python

export PYTHONPATH=$PYTHONPATH:/home/pi/librealsense/build/

export PYTHONPATH=$PYTHONPATH:/home/pi/librealsense

However, it seems to be another problem.

pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: /home/pi/librealsense/build/librealsense2.so.2.24: undefined symbol: __atomic_load_8

CroMocha, 2019-07-08:
If you check out /home/pi/librealsense/build you'll find a few librealsense2.so files(librealsense2.so, librealsense2.so.2.24, ... ), looks like your code chose to import librealsense2.so.2.24 instead of librealsense2.so. Maybe you can try just copying over the correct file.

SirDifferential, 2019-07-10:
Did you by chance get your realsenses working on your raspberry 4? I've been debugging them for a few days and posted my findings in this thread: https://github.com/IntelRealSense/librealsense/issues/4272#issuecomment-508665141

baozinihao321, 2019-07-10:
@SirDifferential  Thanks for your work and suggestion. I will try your methods. And could you explain more about "latomic" (I am not familiar with cmake). Should I add     set(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic") in the CMakeLists.txt? Or Something else? Thanks!

SirDifferential, 2019-07-10:
I didn't try editing the CMake files and instead compiled only one example by calling gcc manually. With Cmake, I guess you need to add `-latomic` to the link flags for all the examples and tools as they link against `librealsense.so`. I think the file you need to edit is `librealsense/examples/CMakeLists.txt` and change this block:

```
if(BUILD_GRAPHICAL_EXAMPLES)
    include(${CMAKE_SOURCE_DIR}/CMake/opengl_config.cmake)
else()
    if(ANDROID_NDK_TOOLCHAIN_INCLUDED)
        find_library(log-lib log)
        set(DEPENDENCIES realsense2 log)
    else()
        set(DEPENDENCIES realsense2)
    endif()
endif()
```

to

```
if(BUILD_GRAPHICAL_EXAMPLES)
    include(${CMAKE_SOURCE_DIR}/CMake/opengl_config.cmake)
else()
    if(ANDROID_NDK_TOOLCHAIN_INCLUDED)
        find_library(log-lib log)
        set(DEPENDENCIES realsense2 log)
    else()
        set(DEPENDENCIES realsense2 -latomic)
    endif()
endif()
```

and if you build the graphical examples, the file is `librealsense/CMake/opengl_config.cmake` and the dependencies must be: `set(DEPENDENCIES realsense2 glfw ${OPENGL_LIBRARIES} -latomic)`

For Python bindings, the file might be `~/librealsense/wrappers/python/CMakeLists.txt`. I wonder if it's enough to just change:

`target_link_libraries(pybackend2 PRIVATE usb ${CMAKE_THREAD_LIBS_INIT})`

to

`target_link_libraries(pybackend2 PRIVATE usb ${CMAKE_THREAD_LIBS_INIT} -latomic)`

baozinihao321, 2019-07-10:
@SirDifferential It works !! after changing the Python binding CMakeLists.txt. Thank you so much!

closed 2019-07-10 by baozinihao321

SirDifferential, 2019-07-10:
If you got your realsense working, I'd like to know if it also works with 1280x720 depth feed. I'm occasionally getting the minimal examples streaming, but it's very unrealiable, and seems to work less often with higher resolution. Often times sending specific commands seems to cut the connection (emitter enable, use preset, get fov...)

baozinihao321, 2019-07-10:
@SirDifferential It seems that I can only use 640x480 depth feed. Maybe the USB3.0 is recognized as USB2.0 and the SDK still use compatible mode.