realsenseai/librealsense #4376: Incompatibility with Windows 10 Pro version 1903

ntsarb, 2019-07-07:
Camera Model                       |  Intel RealSense D435 | 
Firmware Version                   | Signed_Image_UVC_5_11_6_250.bin  | 
Operating System & Version |   Win 10 Professional for Workstations ver. 1903 | 
Intel Device Driver Version    | 6.1.160.14  (dated 1 April 2019) | 
Kernel Version (Linux Only)    |  N/A                                         | 
Platform                                 | PC |
SDK Version                           |  2.24.0.939                          | 
Language                               |  N/A                          | 
Segment			              |  AR                          | 

Intel RealSense Viewer and Depth Quality Tool no longer work after upgrade to Windows 10 Pro version 1903. When I launch the applications, nothing happens.

I was using the software without any issues in 1803, just before Windows Update requested authorisation for installing the 1903 update. On completion of the update, the RealSense software no longer worked.

Remedies attempted:
I uninstalled the SDK and reinstalled. No improvement.

Microsoft Support Case:
I opened a ticket with Microsoft Support, case number: 1476811164
Microsoft's Support recommended me "to contact the manufacturer of your device to get if the have it latest version that is tested to run on Windows 10 1903 or we can go back to the previous build of Windows 10.".

Can you please confirm if this is an incompatibility with Windows 10 Pro 1903 and, if so, consider how to resolve this?

Thank you

divyakush, 2019-07-08:
Hello ntsarb,
You can visit my site and solve your problem manually
I have mentioned a simple and effective process on [open dll file](https://opendllfiles.com/) website.
If you will do not solve your problem then mail me via mail.
thanks 
David Thomas

RealSenseCustomerSupport, 2019-07-23:
----------------------------------------------

Hello ntsarb,
Did you resolve your problem?

ntsarb, 2019-07-31:
> Hello ntsarb,
> Did you resolve your problem?

Thanks for your message. I had stored the camera in a drawer, as it wasn't working. 
I reinstalled SDK Version 2.24.0.939, today, and it looks like it's back on duty.

closed 2019-07-31 by RealSenseCustomerSupport