realsenseai/librealsense #4379: How can I get motion frame such as accel and gyro through frameset from pipeline?

xhuan28, 2019-07-08:
I would like to get motion frames such as accel and gyro frame through frameset from pipleine. However, it seems that there is no such kind of API(get_accel_frame() or get_gyro_frame()) provided by the class of frameset, just like get_depth_frame(), get_color_frame(), get_fisheye_frame(), get_pose_frame(), etc.

eMrazSVK, 2019-07-08:
Hello,

You have all those in pose frame (rs2::pose_frame). At least accel values. Take a look [here](https://github.com/IntelRealSense/librealsense/blob/master/examples/pose-predict/rs-pose-predict.cpp). Especially line#32 in predict_pose function. You can access velocity, acceleration, angular velocity and angular acceleration and of course translation and rotation.

xhuan28, 2019-07-09:
Thanks for your suggestion. Yes, I can get accel and gyro info from pose frame with T265. However, I am thinking of how should I get the information from D435i camera on which there is no pose frame.

eMrazSVK, 2019-07-09:
What about [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#api)? There is clearly a method `get_motion_data()`.

RealSenseCustomerSupport, 2019-07-23:
----------------------------------------------

Hello **xhuan28**
Do you resolve your issue?

rosesue, 2019-07-29:
Hi, @RealSense-Customer-Engineering 
Thanks for @eMrazSVK 's answers. However, I still cannot obtain either pose frame through Frameset::get_pose_frame() or motion frame through Frameset::first_or_default(). It seems that these two functions don't work or this issue is related with timing? Finally, I managed to get the motion data in the way that is something like the following code:
`if (frame.is<rs2::frameset>()) { `
  ` get_color_frame();  `
   `get_depth_frame();  `
  ` ...... ` 
`} else if (frame.is<rs2::motion_frame>()) {  `
  `get_motion_data();  `
`} else if (frame.is<rs2::pose_frame>() {  `
  `get_pose_data(); ` 
`}`  
It seems that it only supports to get video frame data from frameset. Is that a issue? Is there a plan to support get_motion_frame() in Frameset class in future?

xhuan28, 2019-07-29:
Hi, @RealSense-Customer-Engineering 
Thanks for @eMrazSVK 's answers. However, I still cannot obtain either pose frame through Frameset::get_pose_frame() or motion frame through Frameset::first_or_default(). It seems that these two functions don't work or this issue is related with timing? Finally, I managed to get the motion data in the way that is something like the following code:
`if (frame.is<rs2::frameset>()) { `
  ` get_color_frame();  `
   `get_depth_frame();  `
  ` ...... ` 
`} else if (frame.is<rs2::motion_frame>()) {  `
  `get_motion_data();  `
`} else if (frame.is<rs2::pose_frame>() {  `
  `get_pose_data(); ` 
`}`  
It seems that it only supports to get video frame data from frameset. Is that a issue? Is there a plan to support get_motion_frame() in Frameset class in future?

ev-mp, 2019-07-29:
@xhuan28 , the above code is incorrect - you should query the motion frames from the original frameset. Move the two `else` clauses into the first `if (frame.is<rs2::frameset>()) { ` section.
See [example](https://github.com/IntelRealSense/librealsense/blob/f112ed78e3917df79047254f864157195d4488b0/doc/t265.md#api)

ev-mp, 2019-08-05:
@xhuan28 , did you manage to resolve it?

xhuan28, 2019-08-07:
However,  the accel, gyro and pose frame cannot be obtained in the way showed in the above example. The only difference is I was using a callback to start pipeline but not waiting for frames.

closed 2019-08-14 by RealSenseCustomerSupport