realsenseai/librealsense #4381: Nvidia TX2, pipe.start() assertion failed

ambersun1234, 2019-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04.2 LTS (Bionic Beaver)  | 
| Kernel Version (Linux Only)    |  4.9.14.-tegra                                         | 
| Platform                                 | NVIDIA Jetson tx2  |
| SDK Version                            |  2.24.0(Build from source)                          | 
| Language                            |  python                          | 
| Segment			| others                          | 

### Issue Description
I had build the source code from github( f112ed7 ), and successfully installed on tx2, using the following command
```
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build && cd build
cmake -D BUILD_PYTHON_BINDINGS=bool:true ..
make -j`nproc`
sudo make install -j`nproc`
```

When I use pyrealsense2 in python, something weird happened. And by the way i had put `librealsense2.so` & `pyrealsense2.cpython-36m-aarch64-linux-gnu.so` under the same working directory. Part of my python code is as follows
```
import pyrealsense2 as rs
pipe = ts.pipeline()
profile = pipe.start()
```
It works fine in the first 2 lines, but when it runs to the third line, I got this error
```
python3: librealsense/src/pipline/pipeline.cpp:84: 
void librealsense::pipeline::pipeline::unsafe_start(std::shared_ptr<librealsense::pipeline::config>): 
Assertion `profile->_multistream.get_profiles().siz() > 0' failed.
Aborted (core dumped)
```
Has someone run into the same problem?
Need some help guys!

eMrazSVK, 2019-07-08:
Hi,

You have a typo at `pipe = ts.pipeline()` so its not `ts`, its `rs`, so change it to: `pipe = rs.pipeline()`

ambersun1234, 2019-07-08:
@eMrazSVK 
It is just a typo of the article. I've updated the article.

eMrazSVK, 2019-07-08:
Maybe try to look at this #1091
Does your device get recognized as an USB3 device? You can check with `dmesg` command.

ambersun1234, 2019-07-09:
@eMrazSVK 
Result from `dmesg` is as follows
```
[  521.694928] usb 2-1: New USB device found, idVendor=8086, idProduct=0b3a
[  521.694973] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  521.694988] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  521.769346] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  521.780799] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  521.790553] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[  521.798602] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  521.808900] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[  521.816943] input: Intel(R) RealSense(TM) Depth Ca as /devices/3530000.xhci/usb2/2-1/2-1:1.0/input/input5
```
Not sure how to extract some useful information from `dmesg`, so I just copy some lines that looks improtant. I cannot tell whether the device get recognized as USB3 device or not, so I tried [another method](https://askubuntu.com/questions/604158/how-do-i-tell-if-a-usb-thumb-drive-is-usb-3-0). 
I get this:
```
Couldn't open device, some information will be missing
  bcdUSB               3.20
  bcdDevice           50.ad
        bcdUVC               1.50
        bcdUVC               1.50
          bcdHID               1.10
```
According to the [article](https://askubuntu.com/questions/604158/how-do-i-tell-if-a-usb-thumb-drive-is-usb-3-0), it is clear that realsense get recognized as USB3 device.

Could this type of issue related with realsense's firmware?

eMrazSVK, 2019-07-09:
@ambersun1234 You can see in the dmesg (after reconnecting the camera) something like New SuperSpeed device. High Speed is USB2 and Super Speed is for USB3. That's how you can find out. But method you used seems legit. Did you use some kind of USB hub or are you directly connecting it to Jetson? Are you using some kind of 3rd party carrier board? Try changing the cable maybe. But thats just some suggestions I would try at first.

ambersun1234, 2019-07-09:
@eMrazSVK 
I connect `realsense` and `tx2` via the cable which is provided along with realsense. I didn't use any 3rd party carrier board.

The only thing I have connected is just `realsense` and `tx2`.
As for the software part, I only install `opencv` from source, and update `tx2` via `jetpack`.
Not sure those operations will effect `realsense` or not.

I've got another `realsense` here, so maybe I'll just changing the camera and the cable then do AB test to find out the reason instead, and maybe update `realsense's firmware` if nothing changed.

ev-mp, 2019-07-09:
@ambersun1234 hello,
Based on the input the SDK may not recognize the connected realsense devices:
>Assertion `profile->_multistream.get_profiles().siz() > 0' failed.

Please perform the following:
- Verify that you have the udev rules installed [as required](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites):
>./scripts/setup_udev_rules.sh

- Build and run SDK's `rs-enumerate-devices` tool and share the output.

ambersun1234, 2019-07-10:
@ev-mp 
I've run `./scripts/setup_udev_rules.sh`
When i run `rs-enumerate-devices`
it shows
> No device detected. Is it plugged in?

I'm pretty sure that realsense is connected to tx2 via cable
and tx2 did recognized it as USB3 via `lsusb`

ambersun1234, 2019-07-10:
@ev-mp 
I've run `./scripts/setup_udev_rules.sh`
`rs-enumerate-devices` result is as follows
```
Device info: 
    Name                          : 	Intel RealSense USB2
    Serial Number                 : 	843112073092
    Firmware Version              : 	05.10.13.00
    Recommended Firmware Version  : 	05.11.06.250
    Physical Port                 : 	/sys/devices/3530000.xhci/usb1/1-2/1-2:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD6
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	D400
    Asic Serial Number            : 	903223053275

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  1280x720	@ 4Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x480	@ 4Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 4Hz	   Y8
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  1280x720	@ 4Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  640x480	@ 4Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  480x270	@ 4Hz	   Z16

```
I connect `realsense` with `tx2` via provided cable on USB3.0, but it seems that `realsense` get recognized as USB2.0.
By the way, I don't have extra `realsese` for AB test.
Should I update `realsense`'s firmware? Because I found some case that may cause error.
Thanks.

ambersun1234, 2019-07-15:
For someone who encounter with the same issue.

My `realsense` have the firmware version with `5.10.13`, I tried to change the firmware down to `5.10.03`, and everything work like a charm.

closed 2019-07-15 by ambersun1234