realsenseai/librealsense #4382: Ply file giving wrong output

rupeshchandgude, 2019-07-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | RealSense Viewer | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    | 4.15.0-54-generic  | 
| Platform                                 | PC |
| Language                            |  python        | 

### Issue Description
From Realsense viewer I had stored recording as .bag file. From .bag file I generate .ply file for each frame in that video , also I have depth image fro that frame. While I try to visualize ply file using open3d library, output is not relevant. 
Code to convert .bag to .ply
[temp.txt](https://github.com/IntelRealSense/librealsense/files/3367630/temp.txt)

From the above code I got .ply file and depth image for corresponding frame.
To visualize the ply file I use following code:
[temp2.txt](https://github.com/IntelRealSense/librealsense/files/3367635/temp2.txt)
 
Output of ply file:
[image]

Depth image:
[image]

Original image:
[image]

a17sans, 2019-07-08:
I have exactly the same issue for days ... I tried to generate the .ply directly without saving it to a bag file, same results ...

eMrazSVK, 2019-07-08:
From the `output of ply file` isn't it just zoomed out? Did you try to zoom in? I had this issue sometime in the past (I do not remember exact case). Because sometimes it returns points which are too far from the actual captured scene and viewers in general get confused, they try to keep each point in view and so it is zoomed out. If this is the case, you can try to cut the distant points by some kind of filter.

rupeshchandgude, 2019-07-08:
> I have exactly the same issue for days ... I tried to generate the .ply directly without saving it to a bag file, same results ...Thanks for your reply. Can you please tell me how can we save ply file directly from realsense-viewer?

rupeshchandgude, 2019-07-08:
> I have exactly the same issue for days ... I tried to generate the .ply directly without saving it to a bag file, same results ...

Thanks for reply. But I don't know how to generate ply file directly from realsense-viewer. Can you please give reference or code if any?

rupeshchandgude, 2019-07-08:
> From the `output of ply file` isn't it just zoomed out? Did you try to zoom in? I had this issue sometime in the past (I do not remember exact case). Because sometimes it returns points which are [...]

Thanks for answer. After zooming in, I can see the shapes similar to that are in original figures. But they are very noisy. It is very hard to recognize. Can you share any code snippets regarding to cutting out points based on distance?

eMrazSVK, 2019-07-08:
@rupeshchandgude You could use [this](http://docs.ros.org/kinetic/api/librealsense2/html/rs-pcl_8cpp_source.html) as some kind of guide. It uses PCL library to apply the PassThrough filter.

a17sans, 2019-07-08:
> Thanks for reply. But I don't know how to generate ply file directly from realsense-viewer. Can you please give reference or code if any?

Actually I do exactly the same thing that you do in your temp.txt but from the camera directly without using .bag just like this example : [export_ply](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py)

But I have also tried to export a ply from a .bag and I have exactly the same results ... 
Do you concatenate all the .ply you have to get a good .ply ? I'd like to get (in 3D) on the .ply the object I record, and thus I wonder if it would be more accurate to concatenate all the .ply files.

eMrazSVK, 2019-07-08:
Anyways, I found my old work. [Here](https://github.com/eMrazSVK/JetsonSLAM/blob/develop/lib/RS_PCL_converter.cpp) is how you convert RS frame to PCL cloud. Then take a look [here](https://github.com/eMrazSVK/JetsonSLAM/blob/develop/lib/jetSlam_registration_toolkit.cpp#L11) so you see how to use PCL PassThrough filter. Hope it helps.

rupeshchandgude, 2019-07-09:
I have solved the issue. There is small bug in ply file generation code.
I just comment out below line in temp.txt file. 
`config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)`
Which result into correct output. Thanks for your help

closed 2019-07-09 by rupeshchandgude

a17sans, 2019-07-09:
> I have solved the issue. There is small bug in ply file generation code.
> I just comment out below line in temp.txt file.
> `config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)`
> Which result into correct output. Thanks for your help

Did you set particular parameters for your camera ? Will you try to get only one ply file from your .bag ? 

Thanks

rupeshchandgude, 2019-07-12:
> Did you set particular parameters for your camera ? Will you try to get only one ply file from your .bag ?
>
> Thanks

I use default parameter for camera and only extract 1st frame from bag file.