realsenseai/librealsense #4383: Cameras non-responsive after applying PointCloud filter

jb455, 2019-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | Various including latest (05.11.06.250) | 
| Operating System & Version |   Win10 1803 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.21                         | 
| Language                            |  C#                    | 
| Segment			|  other                 | 

Hi,
Over the last few weeks/months we've been having an issue which has been putting cameras in a temporary 'bricked' state.

The exception we get is when calling `var points = (Points)frame.DepthFrame.ApplyFilter(pc);`. Most of the time this works fine, but occasionally (from 1 in 20 to 1 in 50 maybe) it throws `System.InvalidCastException: Unable to cast object of type 'Intel.RealSense.DepthFrame' to type 'Intel.Realsense.Points'` and the camera dies.
When the camera has entered this state, it isn't accessible by our application any more, doesn't show up in RealSense Viewer, and doesn't even appear in Device Manager. Windows sometimes shows a toast saying words to the effect of "The last USB device has malfunctioned" just after the exception is caught. 
With normal D400 cameras it is usually ok if you unplug and plug it back in, but we are also seeing this issue on tablets with integrated D410 cameras, where we've found that you need to do a full shut down (not restart), turn the tablet back on, then leave it for half an hour or so before you can use the camera again. Obviously none of this is ideal when our customers get their hands on the devices.
As far as I can tell we're doing everything correctly when it comes to disposing of objects etc. Is there anything in particular I should look out for in case this is caused by our code and not hardware etc?

Thanks

ev-mp, 2019-07-09:
@jb455 hello,
Please start with adding input check into the code as currently it just assumes that `frame.DepthFrame` actually constitutes a depth frame. The same would apply to the result of `frame.DepthFrame.ApplyFilter(pc)` as the casting does not verify the underlying type. 

Regarding the device's removal from the device manager - can you provide Windows system logs from the integrated D410 cameras and Device Manager's screenshots when this situation occurs ? 
You can also share this info via the ZenDesk channel and refer to this ticket for cross-reference

jb455, 2019-07-09:
Hi, thanks for your reply. I've changed the code like so:
```
if (frame.DepthFrame is DepthFrame)
{
	var filtered = frame.DepthFrame.ApplyFilter(pc).DisposeWith(frame);
	if (filtered is Points)
	{
		points = (Points)filtered;                        
		...
	}
	else
	{
		throw new Exception("Points not valid");
	}
}
else
{
	throw new Exception("Depth frame not valid");
}
```
It throws at "Points not valid" and the camera is still dying.
I don't have one of the other devices to hand at the moment, but when I do, what exactly is it you want? Relevant stuff from the Windows Event Viewer?

jb455, 2019-07-11:
I've tried twice to get to zendesk to submit this with no luck...
Here's the log from a few seconds before the crash, then trying to restart the camera afterwards: https://pastebin.com/P8YfM4Gv
The only thing I could see in the event viewer that looked vaguely relevant from around the same time was a LiveKernelEvent which mentioned USBHUB3 with the signature:
```
Problem signature:
P1: 144
P2: 3003
P3: ffff9b83e305f5c0
P4: 40010001
P5: 0
P6: 10_0_17763
P7: 0_0
P8: 256_1
P9: 
P10: 
```
I can share the full event privately if it's any use.

Screenshot of device manager showing the D410 has disappeared from the Cameras section and Unknown USB Device has appeared (though I'm pretty sure I checked before and there was no unknown device listed other times this issue has occurred):
[image]

closed 2019-07-30 by RealSenseCustomerSupport