realsenseai/librealsense #4384: IMU on Android

cottev, 2019-07-08:
Hi,

I would like to know when is planned the support of the IMU for the camera D435i on Android ?

Best Regards,
Stephane

ev-mp, 2019-07-09:
@cottev hello,
Delivering IMU capability to android platform is a scheduled feature. I'm sorry but I am not able to provide further details at this time.
Please follow the (bi-weekly) SDK releases and notifications.

cottev, 2019-07-09:
Hi,

Thanks for this reply but could you tell me if it is a matter of weeks, months or years ? 
We would need this info to choose the approach of our algo based on your camera.

Br,
Stephane

YunfanZhang42, 2019-07-10:
Following up on this issue here. Our team also needs this feature and would be interested in knowing the release date for IMU support on Android.

kevin2431, 2019-07-16:
@dorodnic Our team are also focused on this issue. We are now developing an app on android which needs imu data for VIO. So it would be very important to know how long can you deal with this problem on android. If this can not be done in the next few months, we need to change the hardware.

yujingxin, 2019-07-17:
Same Problem. 
I am interesting about the support of IMU on Android. I would like to know when you can release it.

qiangz520, 2019-07-17:
For this problem, I am also head-scratching, waiting for its new progress.

zijinxuxu, 2019-07-19:
Using your product, same problem, please release ASAP. Thx

cottev, 2019-07-26:
Would it be possible to Intel to comment on this topic ?
Br,
Stephane

cottev, 2019-07-31:
@dorodnic. Please, would it be possible for you to comment on this issue?
Best Regards, stephane

dorodnic, 2019-08-08:
Hi @cottev 
Sorry for the delay. We cannot comment on features then there is significant technical risk for their schedule. That said, we do prioritize features requested by the community - the team has been working on this for last couple of months, it looks good,  now its safe to say it will be merged very soon (week or two), and you can try it right now by pulling #4605

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

Hi @cottev

Do let us know your feedback based on testing the PR #4605 outlined by dorodnic.

Thanks

cottev, 2019-08-26:
Hi,
Sorry for the delay I was in vacation ;-)
It is a great news and I will take a look to this release.
Thanks again for your effort.
Br,
Stephane

cottev, 2019-08-26:
Hi,
I have just tested with the release of SDK 2.26.0 with an update of the FW camera ( Signed_Image_UVC_5_11_11_100.bin ).
I can us the gyro and the accel if I enable through the following commands:

_config.enable_stream(RS2_STREAM_GYRO);
config.enable_stream(RS2_STREAM_ACCEL);_

However as soon as I try to enable the pose the function pipe.start is blocked.

My code of enabling is quite simple:

_config.enable_stream(RS2_STREAM_DEPTH, WIDTH, HEIGHT, RS2_FORMAT_Z16, FPS);
config.enable_stream(RS2_STREAM_COLOR, WIDTH, HEIGHT, RS2_FORMAT_RGB8, FPS);
config.enable_stream(RS2_STREAM_POSE);
pipe.start(config);_

I have the following message under the logcat on android:
2019-08-26 16:25:29.730 17935-22495/? I/librs: Found HID device: id- gyro_3d
    vid- �
    pid- :
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
2019-08-26 16:25:29.730 17935-22495/? I/librs: Found HID device: id- accel_3d
    vid- �
    pid- :
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
2019-08-26 16:25:29.730 17935-22495/? I/librs: Found HID device: id- custom
    vid- �
    pid- :
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
2019-08-26 16:25:29.731 17935-22495/? I/librs: Found 302 unreachable streams, 302 extrinsics deleted
2019-08-26 16:25:29.783 17935-22495/? W/librs: hwmon command 0x4f failed. Error type: No data to return (-21).
2019-08-26 16:25:29.783 17935-22495/? I/librs: IMU EEPROM Read error: hwmon command 0x4f failed. Error type: No data to return (-21).
2019-08-26 16:25:29.783 17935-22495/? I/librs: Motion Module - no intrinsic calibration, Depth Module V2 intrinsic invalidated : Accel !
**2019-08-26 16:25:29.788 17935-22495/? W/librs: null pointer passed for argument "profile"**
2019-08-26 16:25:29.788 17935-22495/? E/librs DeviceWatcher: Couldn't resolve requests
2019-08-26 16:25:29.788 17935-22495/? D/librs DeviceWatcher: Device: /dev/bus/usb/002/002 added successfully

cottev, 2019-08-27:
Hi,
In fact it is working as the pose is not supported by the camera D435i.
Thanks again,
Stephane

closed 2019-09-03 by dorodnic

mei123hao, 2019-10-21:
The sensor that i use is D435i. Now I want to extract imu  data on android But I don't know how to do. Who can tell me. Thanks a lot !

liubamboo, 2020-12-25:
@cottev Hello, and now can you get the imu data? Could you share the approach to get it?