realsenseai/librealsense #4385: Creating Matlab mex files with realsense library

rcohe032, 2019-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.11.06.250) | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.21.0 }                          | 
| Language                            |  {matlab }                          | 
| Segment			|                          | 

### Issue Description

I am trying to create a mex file from c++ to use in matlab. I am doing this since matlab does not support advance mode. I have created a code in c++ that sets the advanced mode features, and I am making this into a mex file so that I can run this "set-up" in Matlab and then proceed with the rest of my Matlab code. 

My workaround currently is plugging in my D415, opening the RealSense Viewer, loading my .json file, closing RealSense Viewer, and then running my Matlab script (the advanced settings stay loaded on the camera). I would like to eliminate this step by loading the settings in C++ automatically. 

I have my code (attached), and when I try to mex it, it cannot find the librealsense hpp's. I have tried moving the hpp to the cd and adding the librealsense path to matlab. I get this error:

`Error using mex
P:\test mex\test mex\RS_setup.cpp:10:33: fatal error: librealsense2/rs.hpp: No such file or directory
compilation terminated.`

Anyone know how to fix this or a workaround? Thanks! 

[RS_setup.txt](https://github.com/IntelRealSense/librealsense/files/3368910/RS_setup.txt)

lramati, 2019-08-20:
Is librealsense installed in such a way that it's headers are on the default path for finding includes during compilation? What happens if you switch to a relative import for `librealsense2/rs.hpp` as well?

rcohe032, 2019-08-20:
Do you mean the matlab path? or in c++?

My librealsense is installed so that in c++, all the headers are found when I attach the property sheet. Do you mean that in the side bar under headers, all of the headers should appear? I have also manually added the hpp's to the sidebar in visual studio under Headers. After that didnt work i copied all of teh hpp's into the folder where my sln and cpp is. 

When i move my RS_setup.cpp to the include folder in the realsense SDK in program files, it says I dont have access. 

What do you mean by switch to relative import? 
Thanks!

lramati, 2019-08-20:
I was wondering why you've got `<realsense2/rs.hpp>` but also `"../include/librealsense2/rs_advanced_mode.hpp"` (Note the difference between <> and "")

lramati, 2019-08-20:
It seems like mex is failing to find the header, which is why I'm struggling to help you. Have you tried duplicating the current librealsense_mex project inside the solution generated by CMake and re-purposing that for your mex file? That could help make sure things are properly configured

rcohe032, 2019-08-20:
Thanks @lramati 

I dont really know why the two are different (probably copied from two different other codes) but that has been working fine when i run code in c++. (no issues)

Could you give me some guidance on how to repurpose the mex file? Is it readable in any application? I think the issue is that the librealsense_mex doesnt have the advanced controls in it, so I am not sure how to repurpose it or even open it outside of the matlab mex files already made. 

I just tried copying the entire include folder from the realsense sdk into the project folder so it could find the headers and now it can find those. But it gives a bunch more errors that it cannot find all the other files. 

 example: AppData\Local\Temp\mex_6174479058758884_12556\RS_setup.obj:RS_setup.cpp:(.text$_ZN11MexFunction7setupRSEv[_ZN11MexFunction7setupRSEv]+0x347): undefined
reference to `rs2_set_depth_table'

If I could just tell matlab/the c++ code where to look for the librealsense library in c++ it should work? But maybe it's not that simple.

Alternatively, do you know if there will be a release soon where we will be able to adjust advanced controls in matlab? 

Thanks so much for your help!

lramati, 2019-08-20:
I meant reuse the entire visual studio project. Like, delete all of the
prexisting files from inside of it, and then add yours instead. That would
effectively "transfer" the settings from the current Mex file which we know
compiles to your project, hopefully allowing it to compile.

Adding advanced mode to the official wrapper is definitely on the list,
unfortunately I don't have an estimate on when that will be completed.

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

@rcohe032 Any other questions about this? Looking forward to your reply.Thanks!

rcohe032, 2019-09-05:
I still wasnt able to figure it out unfortunately. Since it is not a big deal for me I have just been using the workaround of putting pauses in my matlab code and changing the disparity manually in the RealSense Viewer.

Thanks anyways for your help