realsenseai/librealsense #4387: Frame Rate

HiOtto, 2019-07-09:
Happy to help. 

To enable IMU explicitly, you can call `config.enable_stream(RS2_STREAM_ACCEL);` and `config.enable_stream(RS2_STREAM_GYRO);`.

You could also specify the frame-rate using `config.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 250)` (this will set the accelerometer rate at 250 Hz)

You can see the list of supported configurations [here](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-motion.h#L49-L53) and in the device data-sheet. 

We don't include explicit configuration details in the examples, because we want the same examples to work on all supported (and future) devices. So if you don't provide an explicit configuration, you will just get the default IMU from the D435i.

Once you do get a frameset with motion frames, you can cast them to [rs2::motion_frame](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L772-L798):
```cpp
using namespace std;
using namespace rs2;
frameset fs = p.wait_for_frames();
for (auto& f : fs) {
   if (f.is<motion_frame>()) {
       motion_frame mf = f.as<motion_frame>();
       rs2_vector xyz = mf.get_motion_data();
       cout << f.get_profile().stream_type() << ": " 
            << xyz.x << ", " << xyz.y << ", " << xyz.z << endl;
   }
}
```

_Originally posted by @dorodnic in https://github.com/IntelRealSense/librealsense/issues/2945#issuecomment-449555411_

1. How you align the 250/400Hz imu data with15/30/60/90 Hz image data when 15N can not be exactly divisible by 250/400 ?
2. Is it possible to run the rgb/depth sensors at another framerate while triggered by external pulse (e.g. 20/40 fps, instead of 15/30/60/90) ?

ev-mp, 2019-07-09:
@HiOtto hello,
1. Librealnsense SDK provides a syncer class utility that aligns the frames temporally relying on stream-specific heuristics and the different timestamps provided. The syncer tries to minimize the anticipated latency while delivering temporally-coherent framesets. There are certain cases, such as synchronizing Depth and IMU, where the basic fps rate differ by an order of magnitude. In those cases the syncer may select a different approach and deliver the high-fps stream data directly while queuing and synchronizing the "low-fps" streams in user--friendly manner.

2. This seem to be counter-intuitive to activate both the HW sync and also try to stream with different fps rates as these seem to be mutually exclusive use-cases. Can you please elaborated what is the actual scenario for this setup?

HiOtto, 2019-07-09:
> @HiOtto hello,
>
> 1. Librealnsense SDK provides a syncer class utility that aligns the frames temporally relying on stream-specific heuristics and the different timestamps provided. The syncer tries [...]

1. "minimize the anticipated latency" means there is an nonconstant latency between the imu timestamp and the image/depth timestamp provide by the driver ? 
If the imu is f1 Hz and image is f2 Hz, with f1 >f2, the latency is supposed to vary between +-1/(f1) s ?
2. I want to sync the image/depth data with a external pulse signal without using imu data of realsense, the pulse is from the system clock of another device, which is should be 2KHz. I want to avoid the timestamp misaligment caused by the non-divisible , so is it possible to trigger the realsense rgb/ir camera with an INT Hz external pulse signal that can be divisible with 2000? like 20/40Hz

ev-mp, 2019-07-12:
@HiOtto hello,
The syncer's first priority is to synchronize Depth with the corresponding 2D image (RGB, IR or FishEye) which are required for image registration, alignment and point cloud scenarios.
So in case of synchronizing UBC streams with zero frame drops and the processing does not introduce additional delay, the expected latency on average should lean toward 1/2 period of the stream with the lower fps.  (16.6 msec in case of 30Hz)
When using IMU streams the pipeline's syncer will not try to synchronize those against the slower Depth/RGB streams but rather push them to the user immediately.
2 Currently, if the camera is configured as a slave device and receives a trigger at a rate different than its preset, then it will fall back and switch to the internal timer to generate frame at the pre-configured fps (e.g 15/30/60/etc')

closed 2019-07-29 by ev-mp