realsenseai/librealsense #4388: Using librealsense as external library in Ubuntu

hosseinpg, 2019-07-09:
I have a D435 camera and I'm trying to use librealsense library as external source. Unfortunately, I can not find related ...config.cmake file or the related module. Are they provided? If not how can I make use of librealsense as external library?

ev-mp, 2019-07-09:
@hosseinpg hello,
The required cmake configuration files for external compilation and linkage are deployed via `sudo make install` or `sudo apt install librealsense2-dev` commands depending on the build/deployment schemes selected.

Adding librealsense2 to external cmake as a build dependency then can be done with the standard CMake notation. See the comment by @radfordi [here](https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118)

hosseinpg, 2019-07-09:
Thank you for your replay dear @ev-mp. I'm a little bit confused. I have installed the realsense peckage using the description provided in this link: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
In this case I have tested the viewer and it is working. Also my projects cmake file can find the realsense2 package using the find_package function. But I could not use gfw3 for example, so I cloned the repository trying to access those files. Now, I have the repository but I don't know how to use it's resources in my project. I hope my problem is clear. 
Thank you.

radfordi, 2019-07-09:
Hi @hosseinpg, you likely just need the GLFW headers.  In case you don't have the GLFW or `librealsense` headers, can get them with:
```sh
   sudo apt-get install libglfw3-dev
   sudo apt-get install librealsense2-dev
```
Then, in your `CMakeLists.txt` file you would do something like this:
```cmake
find_package(glfw3)
find_package(realsense2)

target_link_libraries(${PROJECT_NAME} PRIVATE realsense2::realsense2 glfw)
```
You might also need OpenGL, so you'd add:
```cmake
find_package(OpenGL)

target_link_libraries(${PROJECT_NAME} PRIVATE OpenGL::OpenGL)
```

hosseinpg, 2019-07-15:
@radfordi Thank you. Now I have another related problem when I'm trying to make.
[100%] Linking CXX executable scanner
/usr/bin/ld: cannot find -lrealsense2
collect2: error: ld returned 1 exit status

I would appreciate if you could help.

radfordi, 2019-07-16:
At this point, I suspect your issue is the same as #4410.

hosseinpg, 2019-07-27:
@radfordi Thank you dear Jim. I'm having a version mismatch error and trying to solve it.

closed 2019-07-27 by hosseinpg