realsenseai/librealsense #4390: Missing points on tube borber

descamps2, 2019-07-09:
I am trying to measure a tube diameter using a D415, but i see a lot of missing point on the tube border. The D415 is placed parallel to the tube orientation (see the image attached from realsense viewer), so there should be little occlusion, even on the borders. 

I understand that the surface on the border of the tube can be more difficult to detect, because the reflection from the laser projector will be weaker. However, i observe that the missing points are mostly on one side of the tube. If i reverse the orientation of the sensor, the side of the missing point changes, so that is not an illumination problem. Moreover, the behaviour changes depending of the background behind the tube, it is much better with bright background.

Do you have any idea where this comes from, and what can i do to solve or mitigate this issue?

Thanks,

Adrien

Camera Model  : D415
Firmware Version  : 
Operating System & Version  : Ubuntu 18.04  
Kernel Version (Linux Only) :  4.15.0 
Platform : PC
SDK Version : 2.23 

[image]

AntoineWefit, 2019-07-09:
If you haven't already, I would advise you to tune your camera's parameters, those are saved in a configuration file (.json).
This can be done with the Intel RealSense Viewer. You'll then need to apply these parameters before streaming.
Tell us if you need more details about that !

I'm very confident that a finely tuned camera will give you much better results than what you've shown :)

descamps2, 2019-07-10:
Depending on the illumination, i can most of the time get better results than the images i have shown, that was mostly to illustrate the problem.
For the parameter tuning, i have tried to tune the exposure and auto exposure (most important, i need to increase the mean intensity set point for best result), the projector power (important indoor, much less outdoor), and the various presets for depth calculation (the best seems to be the default). I have not tried to tune the depth calculation parameters individually.
However, i have difficulties finding a setting that works reliably, knowing that it should work outdoor in a large variety of illumination condition.

My question was mostly about the asymmetry of the result (much better result on the bottom side than on the top side of the depth image), because i see no reason for this behavior.

AntoineWefit, 2019-07-10:
I understand your concern, it is indeed tricky to find a set of settings that works for a variety of conditions. Lighting conditions are a big deal with these cameras, especially when we are talking about variable conditions.
I guess that trying to tune the auto-exposure is a good start. If you're willing to do so, you could also add some physical filters on your camera. Try [these long-pass IR filters](https://www.edmundoptics.fr/p/1quot-diameter-optical-cast-plastic-ir-longpass-filter/5417/), they greatly improve the mesh quality in noisy conditions. They are cheap and simple, but if you are really interested, you should buy [these much more efficient ones](https://www.edmundoptics.fr/p/850nm-cwl-10nm-fwhm-125mm-mounted-diameter/22452/).

You are talking about 
> a large variety of illumination condition

But what is your typical working distance ? Is it fixed ? The camera accuracy depends on the distance (the closer, the more accurate). I know that it is possible to find a set of parameters working well for a given distance.

Hope this'll help !

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@AntoineWefit Thanks very much for your input on this.
@descamps2 Sorry for the late reply. Did you get through the issue now? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@descamps2 Could you please update if you need any more support on this? Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@descamps2 If no other questions for this ticket, will close it at this point. Thanks!

closed 2019-10-12 by RealSenseCustomerSupport