realsenseai/librealsense #4392: Launching mavros px4.launch causes Realsense T265 '/camera/odom/sample' topic's pose to drift significantly

imoss15, 2019-07-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  (4.14)                                         | 
| Platform                                 | Odroid XU4  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
For our project we use a pixhawk mini flight controller with mavros px4 connected to an Odroid XU4 with WIFI to use most of our setup offboard through ROS. The realsense t265 seems very promising as a plug and play VIO sensor, however we are experiencing a strange issue that I will try to outline as best as possible. 

The realsense works well on its own. We first set it up on Windows to test, then linux to test the realsense-ros wrapper. After validating it as a potentially useful VIO sensor, we managed to get the realsense launching from our Odroid after a couple days of pain figuring out that we needed the 4.14 kernel and that not all nodes would launch unless we flashed a specific ubuntu 16.04. 

After all this, we managed to launch the realsense t265 on the odroid succesfully and with fairly accurate results. We have a motion capture system that is run through a matlab script that synchronizes rigid body pose with the realsense pose and plots them together. 

Our issue however, is that for some reason, the second that mavros px4.launch is launched, the pose data from 'rostopic echo /camera/odom/sample' immediately starts linearly increasing at a fast rate. It appears that the linear twist gets increased as well which could be the cause. My intuition would lead me to believe that it must be something inherent in px4, possibly the launch file or configurations. At first we thought that it could be CPU/RAM related, but we did tests to show that is likely not the case.

I've set the ekf2 to not use vision and tried Attitude Q estimator without vision and we still have the same problem immediately when mavros is launched. I also went through the px4_config.yaml and made all parameters related to vision set to false which also did not fix anything.

For a sanity check we launched mavros and the realsense on a seperate computer which worked fine (why we originally thought it may be CPU/RAM related)

I would appreciate any help and support, please let me know if you require any additional information. We hope to use the realsense VIO to fly autonomously in position hold mode.

MKargus0, 2019-07-13:
I have the same problem when I use  raspbery pi 3b (ubuntu mate 18.04) as companion computer [.](url
[image]
)
With jetson TX2 or laptop (ubuntu 18.04) position and velocity  has been normaly changing.

reyes73a, 2019-07-19:
I see the same issue on RPi 3B+, Jetson Nano and sometimes on Jetson TX1. This is 100% a ros wrapper issue because sdk examples work just fine. Also have of the Ros examples don't work, fail after a while or give errors...

start0821, 2019-07-29:
Hi guys! 
I used RPi 3B+, ubuntu mate 16.04, ros kinetic, px4 1.8.2 on pixhawk1.
And all nodes are processed in RPi, it worked well ! (position data doesn't drift, not increase) 
I used LPE mode of px4, and actually tested a drone in position mode.

Actually I want to process on odroid xu4, because of RPi3 B+ low performance. odroid xu4 has better performance than RPi3 B+, so I don't think cpu is problem.
sorry for bad english :)

imoss15, 2019-07-29:
I believe we were able to fix our issue by changing back parameters within the px4 launch script to false for anything related to vision. At first we thought that the fix was Wifi related because we simply brought the drone from the motion capture back to the lab and were surprised to see that it was working perfectly fine with no drift. It appears rather that the Odroid required a full shutdown for the changes to take affect as now our drone works perfectly fine in both locations with different Wifi sources. I appreciate the collaborative help, perhaps our solution could also be the case for you all? We've had very good performance from the Realsense T265 over the past two weeks. Position Hold flight mode works great and the error is rather minimal even when compared with Motion Capture.

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

Glad to see things are working better now.  Closing out this issue.

closed 2019-08-22 by RealSenseCustomerSupport