realsenseai/librealsense #4393: How is defined T265 confidence ?

BriceRenaudeau, 2019-07-10:
Hello,
I would like to know how is the confidence given by the t265 is defined. 
- Is it related to the position or the motion?
- Is it related to the number of feature in images?
- Is it related to image histogram?
- Is it related to accel/gyro measurements?

In the documentation, I just found an introduction to the T265 with no mention of the confidence.

doisyg, 2019-07-11:
Related to this issue: https://github.com/IntelRealSense/realsense-ros/issues/770

RealSenseCustomerSupport, 2019-07-24:
----------------------------------------------

@radfordi ? @schmidtp1
Any comments?

RealSenseSupport, 2020-09-09:
Thank you for highlighting the need for documenting the details around confidence provided by the T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this in our documentation for T265.

closed 2020-09-16 by RealSenseSupport