realsenseai/librealsense #4397: Run SDK examples from .bag file

maipbui, 2019-07-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.0                        | 
| Language                            | C/C++                       | 

### Issue Description
Hello,

I've tried to run SDK examples from .bag file. I know that there is a way to run using this doc (https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md)
I followed the instructions and modified the code but haven't been able to run the example. What is the correct way to run it? Thank you.

ev-mp, 2019-07-11:
@maipbui hello,
You can load and replay the bag files directly via realsense-viewer.
To see how to load and replay bag files in code check the [rs-record-playback](https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback) tutorial.

maipbui, 2019-07-11:
Hi @ev-mp ,
Thanks for your reply. 
I mean I want to run SDK examples (e.g rs_align, rs_measure etc) from pre-recording *.bag files. Before I always plugged the camera in, run the example. Now I don't have the camera and I want to run SDK examples from my pre-recording *.bag files. How should I do it?

ev-mp, 2019-07-11:
@maipbui, please check tutorial link from the previous response :
The difference between the live vs playback configuration can be narrowed down to the pipe's configuration:

>rs2::config cfg;
    **cfg.enable_device_from_file("a.bag");**
    pipe->start(cfg); //File will be opened in read mode at this point

Note that after modifying the initialization you'll also need to discover which stream types are available in the bag file. To know in advance which streams/configuration are present in the file run `rs-enumerate-devices -p <my_rosbag.bag>`.

maipbui, 2019-07-11:
I got it run as I wanted now.
I need to save RGB stream to analyze some data. Is it possible to convert bag file to video format like `mp4` or `avi`? 
[image]
For example, can I save the color stream of `depth_under_water.bag` directly to my computer with video format? Is there any way to achieve this?

dorodnic, 2019-07-13:
> Is it possible to convert bag file to video format like mp4 or avi?

There is no built-in functionality in librealsense for this, but you can accomplish it with python + pyrealsense2 + opencv2-python script.

closed 2019-07-13 by dorodnic