realsenseai/librealsense #4399: modinfo: ERROR: Module hid_sensor_gyro_3d not found

OOyindamola, 2019-07-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |   Ubuntu Mate armv7l  | 
| Kernel Version (Linux Only)    |  4.4.38                                     | 
| Platform                                 | Raspberry Pi 3  ARMv7 |
| SDK Version                            |  2.0                          | 

Issue Description                                                                                   
--------------------------------------------------------------------------- 

Hi, I followed the instructions in the below link for installing Intel Realsense on Raspberry Pi 3 Model B: https://github.com/IntelRealSense/librealsense/blob/v2.20.0/doc/distribution_linux.md

However, I had two issues.
1.  sudo apt-get install librealsense2-dkms"
This error came up:
"modinfo: ERROR: Module hid_sensor_gryo_3d not found
mod: hid_sensor_gryo_3d
modinfo: ERROR: Module hid_sensor_accel_3d not found
mod: hid_sensor_accel_3d"

2. sudo apt-get install librealsense2-utils"
This error came up:
"E: Unable to locate package librealsense2-utils"

Kindly assist.

ev-mp, 2019-07-11:
@OOyindamola  hello,
The debian distribution currently supports [X86 architecture](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md) PCs.
For ARM-based board check the [Building from Source guide](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md)

OOyindamola, 2019-07-11:
Thanks a lot. 

I reinstalled the Ubuntu Mate 64-bit Version and my kernel version is 4.15. I did not get the error from installing the librealsense2-dkms package as I had earlier but I was not able to install librealsense2-utils. The error is still "unable to locate package librealsense2-utils".

Kindly assist.

ev-mp, 2019-07-12:
@OOyindamola, I'm not sure that I understand  which PC you're trying to install - is it ARM or X86 Based?
To deploy the debian installation on X86/X64 Debian you should follow this [link](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages).
The Non-debian/manual installation link is available in the previous reply.

dorodnic, 2019-07-13:
At the moment [instructions for getting librealsense on the Pi](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md) necessitate building the library from source, you cannot install the library via apt-get.

closed 2019-07-13 by dorodnic