realsenseai/librealsense #4400: .bag  file post-processing

nevynwong, 2019-07-11:
| Camera Model                       | D415 | 
| Firmware Version                   | FW05.11.06.250 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.21 | 
| Language                            | matlab | 

I want to do depth post-processing on all the depth frames that have been recorded in the.bag file, but I don't know what to do?

ev-mp, 2019-07-12:
To use the filters with pre-recorded bag files you can follow the [post-processing](https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing) tutorial and then modify the configuration part of it to switch to a playback rather than a live  device. Also see a relevant 
 discussion in #4397

nevynwong, 2019-07-12:
I have now completed the single frame depth image processing, but I do not know how to take out all the depth frames in the.bag file.

dorodnic, 2019-07-13:
You can skip to next frame(s) using `pipeline.wait_for_frames` just like you would with a real camera.

nevynwong, 2019-07-14:
I used the following code to extract the depth frames in the recorded .bag file and post-process and align, but the number of depth frames obtained is not the same as the number of depth frames extracted by rs-convert.exe. How to deal with it? Can you help me?

pipe = realsense.pipeline;
config = realsense.config;
config.enable_device_from_file('F:\test_bag\848.bag',0);
profile = pipe.start(config);

%%
dec_filter = realsense.decimation_filter();
dec_filter.set_option(realsense.option.filter_magnitude,1);
temp_filter = realsense.temporal_filter();

spa_filter = realsense.spatial_filter();
spa_filter.set_option(realsense.option.filter_magnitude, 5);
spa_filter.set_option(realsense.option.filter_smooth_alpha, 1);
spa_filter.set_option(realsense.option.filter_smooth_delta, 32);
spa_filter.set_option(realsense.option.holes_fill, 3);

fill_filter = realsense.hole_filling_filter();

depth_to_disparity = realsense.disparity_transform(1);
disparity_to_depth = realsense.disparity_transform(0);

depth_sensor = profile.get_device().first('depth_sensor');
scale = depth_sensor.get_depth_scale();%%scale

pointcloud = realsense.pointcloud();
align = realsense.align(realsense.stream.color);

%%playback
playback = profile.get_device().as('playback');
playback.get_duration();
playback.set_real_time(0);
num = 0
while playback.current_status() == realsense.playback_status.playing
% %create a point cloud player
% % Main loop
num = num + 1;

frameset = pipe.wait_for_frames();
colorizer = realsense.colorizer()

depth_frame = frameset.get_depth_frame();
colorized_depth = colorizer.colorize(depth_frame).get_data();

color_frame = frameset.get_color_frame();
color = color_frame.get_data();
img = permute(reshape(color',[3,color_frame.get_width(),color_frame.get_height()]),[3 2 1]);
% Display image
figure
imshow(img);
%%
% player1 = pcplayer([-1 1],[-1 1],[-1 1]);
pointcloud.map_to(color_frame);
points = pointcloud.calculate(depth_frame);

 

%%

frameset = align.process(frameset);
%%
aligned_depth_frame = frameset.get_depth_frame();

frame = aligned_depth_frame;
%%
frame = dec_filter.process(frame);
frame = depth_to_disparity.process(frame);
frame = spa_filter.process(frame);
frame = temp_filter.process(frame);
frame = disparity_to_depth.process(frame);
frame = fill_filter.process(frame);
% Display image
data = colorizer.colorize(frame).get_data();
col_depth = colorizer.colorize(frame);
img = permute(reshape(data',[3,col_depth.get_width(),col_depth.get_height()]),[3 2 1]);
% figure
% imshow(img);

color_frame = frameset.get_color_frame();
colorized_depth = colorizer.colorize(frame).get_data();
col_depth = colorizer.colorize(frame);
% Get actual data and convert into a format imshow can use
% (Color data arrives as [R, G, B, R, G, B, ...] vector)
data = colorized_depth;
img = permute(reshape(data',[3,col_depth.get_width(),col_depth.get_height()]),[3 2 1]);
% Display image
% figure
% imshow(img);

% player2 = pcplayer([-1 1],[-1 1],[-1 1]);
pointcloud.map_to(color_frame);
points = pointcloud.calculate(frame);

end

% Stop streaming

pipe.stop();

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@nevynwong Sorry for late reply. Did you get through the issue? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@nevynwong Do you still need support on this? Thanks!

nevynwong, 2019-09-29:
I have solved this problem. Thank you for your help.

Best Regards!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@nevynwong That's great. Thanks for the update!

closed 2019-10-12 by RealSenseCustomerSupport