realsenseai/librealsense #4405: Can't open the D435 color camera using LabVIEW

eyhtqd, 2019-07-11:
Required Info:
Camera Model - D435
Operating System & Version - Win 10 64bit
Language - LabVIEW 2017 64bit

Objectives:
I would like to open the D435's color camera using LabVIEW

My problem :
I would like to open the D435's color camera using LabVIEW, and I using the 'RealSense_HelloWorld_Depth_and_Color.vi' downloaded from https://github.com/IntelRealSense/librealsense/tree/master/wrappers/labview, but the program alerted me to an error, which said : 
Error 1097 occurred at Call Library Function Node in libRealSense3.lvlib:rs3 open multiple sensor.vi->libRealSense3.lvlib:rs3 start all 2.vi:2930001->RealSense_HelloWorld_Depth_and_Color.vi

Possible reason(s):

LabVIEW:  An exception occurred within the external code called by a Call Library Function Node. The exception might have corrupted the LabVIEW memory. Save any work to a new location and restart LabVIEW.

I located the problem in rs3 open multiple sensor.vi, maybe the realsense2.dll didn't contain the function which can open the color camera? Could you help me solve this problem? I have to use LabVIEW due to work. 
Thanks
Joe

agrunnet, 2019-07-11:
It is because that demo is trying to open the LEFT stereo camera color channel, and not the third RGB sensor. The D435 is only monochrome so can’t open as a color channel.

RealSenseCustomerSupport, 2019-07-29:
----------------------------------------------

Hi eyhtqd, Do you need more help on this?

eyhtqd, 2019-07-30:
> Hi eyhtqd, Do you need more help on this?

Yeah, I still haven't solved the problem. Could you tell me how to solve it?

agrunnet, 2019-07-30:
If you want to stream the color camera then you need to look at one of the other hello world demos. Try to run them all. 

The hello world you chose says it was written for the D415, which uses color sensors for stereo and it therefore opens a “left color camera” which is the stereo camera. I hope that helps.

closed 2024-03-29 by MartyG-RealSense