realsenseai/librealsense #4406: Mutil Intel D415

LITONG12345, 2019-07-11:
Is it possible to call multiple Intel D415 cameras simultaneously in IntelRealSense UnitySdk?
Where can I change in your Demo?
Or do you have any good Suggestions? I am looking forward to your reply

MartyG-RealSense, 2019-07-11:
A while ago, a RealSense community member called frenchtouchXR edited the **RsConfiguration.cs** script of the Unity wrapper to support recording with multiple cameras.  They made their edit beneath the **case mode.Record** instruction at line 42 of the script.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsConfiguration.cs#L42

Directly below case mode.Record, they inserted this instruction:

cfg.EnableDevice(RequestedSerialNumber);

So the end of the script would then look like this:

*********************************************

case Mode.Record:
cfg.EnableDevice(RequestedSerialNumber);

 foreach (var p in Profiles)
                    cfg.EnableStream(p.Stream, p.StreamIndex, p.Width, p.Height, p.Format, p.Framerate);
                if (!String.IsNullOrEmpty(RecordPath))
                    cfg.EnableRecordToFile(RecordPath);
                break;

        }

        return cfg;
    }
}

LITONG12345, 2019-07-15:
Thank you very much for your reply. I will try

closed 2024-03-29 by MartyG-RealSense