realsenseai/librealsense #4407: RealSense error calling rs2

BertrandKevin, 2019-07-11:
Hello everyone, 

I'm using 3 cameras D435 and connected to an UpBoard running Ubuntu 18.04 LTS. The problem is that when I launch my code, sometimes it works and sometimes it didn't (my code is launched with crontab).

When it works, I save 4 pictures and when it didn't work, I receive different message. 

> RealSense error calling rs2_create_device(info_list:0x55f0fdfd8d00, index:2): hwmon command 0x10 failed. Error type:  (1).

> RealSense error calling rs2_pipeline_start_with_config(pipe:0x55bb8617a540, config:0x55bb85f6f240): hwmon command 0x10 failed. Error type:  (1).

The picture bellow shows you the log of my crontab. 
[image]

I also give you the link to my code:
[Code](https://github.com/BertrandKevin/RealSense-Multi-Saving/blob/master/UserFunction/Multi-Camera-Save/src/main.cpp)

Thank you to helping me.

closed 2019-07-12 by BertrandKevin

vhanded, 2019-08-16:
Is this issue solved? I am having same problem.

BertrandKevin, 2019-08-19:
Hello @vhanded,

Unfortunately, no I could solve his issue... The only thing I found is to restart over and over the program. So the program exit failure and restart automatically after that until it works