realsenseai/librealsense #4409: CustomRW on raspbian

kamadhenu87, 2019-07-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D410  | 
| Firmware Version                   | 5.9.11 | 
| Operating System & Version |   Raspbian Debian Stretch 8.0 | 
| Kernel Version (Linux Only)    |  4.14.79-v7+                                        | 
| Platform                                 | Raspberry Pi 3 |
| SDK Version                            | 2.12.0                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### CustomRW
<feature request>
CustomRW software on raspbain to reset to gold/factory calibration.
This would help to reset in remote site with no windows PC access since we have embedded unit using realsense.

dorodnic, 2019-07-13:
Hi @kamadhenu87 
Currently we don't provide calibration stack for non-Intel platforms. This can change in the future (as it did with the firmware update logic, we always want to make as much of code open-source as possible) but it takes time.
Why would you want to be resetting calibration to gold on an embedded system? 
We do plan to release lightweight target-less calibration option as part of librealsense, so stay tuned.
If you insist on setting calibration to gold, we do have this flow somewhere inside librealsense [codebase](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-factory.cpp#L674-L680) as reference.

kamadhenu87, 2019-07-15:
Calibration table is corrupted at some rare occasion and we did gold/factory reset on windows which solved the problem.
So we wish to have the reset application on system as first measure.

kamadhenu87, 2019-07-24:
Hi @dorodnic 
Please can you guide how to compile that portion of code.
https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-factory.cpp#L674-L680

dorodnic, 2019-08-12:
Hi @kamadhenu87 
Sorry for the delay. Restoring factory depth calibration can be done using the following snippet: 
```cpp
rs2::debug_protocol dbg = rs2::context().query_devices().front();

std::vector<uint8_t> restore_def{
        0x14, 0, 0xAB, 0xCD, 0x61, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
};
auto res = dbg.send_and_receive_raw_data(restore_def);
```
Restoring depth to RGB calibration is a bit more complicated.

RealSenseCustomerSupport, 2019-08-30:
----------------------------------------------

@kamadhenu87, if there is no further question, we will go ahead and close this ticket.

closed 2019-09-03 by dorodnic