realsenseai/librealsense #4410: Linux: failed to link with librealsense2

osadeh, 2019-07-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  N/A | 
| Platform                                 | PC   |
| SDK Version                            |  2.24.0 | 
| Language                            | C| 
| Segment			|  N/A| 

### Issue Description
Linking a C app with librealsense2 fails. The reason is that the symbolic link /usr/lib/x86_64-linux-gnu/librealsense2.so links to librealsense2.so.2.23 (which is missing) rather than librealsense2.so.2.24

ev-mp, 2019-07-11:
@osadeh hello,
Make sure to pull the latest v2.24 debian - "sudo apt update && sudo apt upgrade"
Reinstalling is also a viable option : " sudo apt purge librealsense2* && sudo apt install librealsense2-dev llibrealsense2-dbg ..."
Also make sure that you don't have multiple so versions installed, like in "/usr/local/lib/librealsense2*.."

osadeh, 2019-07-11:
I did. I removed and purged all librealsense* packages, and then ran
```
apt update

# I didn't run apt upgrade - too much impact, and I don't think it should make a difference

apt install librealsense2-dev

ls -l /usr/lib/x86_64-linux-gnu/librealsense*
lrwxrwxrwx 1 root root       21 Jun 30 20:36 /usr/lib/x86_64-linux-gnu/librealsense2.so -> librealsense2.so.2.23
lrwxrwxrwx 1 root root       23 Jun 30 20:36 /usr/lib/x86_64-linux-gnu/librealsense2.so.2.24 -> librealsense2.so.2.24.0
-rw-r--r-- 1 root root 16612368 Jun 30 20:36 /usr/lib/x86_64-linux-gnu/librealsense2.so.2.24.0
-rw-r--r-- 1 root root  1202200 Jun 30 20:36 /usr/lib/x86_64-linux-gnu/librealsense-file.a

```

Also tried it on a clean docker image - same results

mepix, 2019-07-11:
I'm having the same issue as well, and tried the same workarounds (purging 2.24.0, reinstalling, updating the udev rules).  It seems like this is related to [Issue 4351](https://github.com/IntelRealSense/librealsense/issues/4351).  Though that issue is marked as "solved", the work around suggested works for building with CMAKE directly and may not work with other popular IDEs like QT.

radfordi, 2019-07-12:
@ev-mp, this is definitely a bug in the `librealsense2-dev` package version `2.24.0-0~realsense0.1244` which has a symlink to the wrong version of the library:
```sh
$ ls -l /usr/lib/x86_64-linux-gnu/librealsense2.so
lrwxrwxrwx 1 root root 21 Jul  1 22:40 /usr/lib/x86_64-linux-gnu/librealsense2.so -> librealsense2.so.2.23
```

hosseinpg, 2019-07-20:
Dear @osadeh 
Have you solved this error? I have the same problem.

osadeh, 2019-07-20:
@hosseinpg, I worked around it:
```
cd /usr/lib/x86_64-linux-gnu/
sudo rm librealsense2.so
sudo ln -s librealsense2.so.2.24 librealsense2.so
```

tarekmuallim, 2019-07-22:
I have the same problem.
I am using Qt Creator.
Any solution?

hosseinpg, 2019-07-27:
Dear @osadeh 
Thank you for your help. It solved the problem. But I faced a API version mismatch. see below error please:      API version mismatch: librealsense.so was compiled with API version 2.24.0 but the application was compiled with 2.23.0! Make sure correct version of the library is installed (make install)

any idea?

ev-mp, 2019-07-28:
@osadeh, @tarekmuallim, the incorrect symlink in the debians should be fixed by now.
Please run `sudo apt update && sudo apt upgrade` and update.

osadeh, 2019-08-05:
Looks good. Thanks for the update
Oren

ev-mp, 2019-08-05:
Thanks for the confirmation

closed 2019-08-05 by ev-mp