realsenseai/librealsense #4411: Error 0x80070005 - D435

Alexander-Shamin, 2019-07-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | FW Version: 5.9.13.0 | 
| Operating System & Version |   Win 10  | 

| SDK Version                            |  2.21, 2.22, 2.23, 2.24                        | 

### Issue Description
Error, then working with intel realsense d435. Install SDK from 2.21 to 2.24. Work with prev version SDK, but i need 2.21 for nuitrack.
~~~ Backend in rs2_create_device(info_list:000002B4404C4530, index:0):
 MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "B:070=> 2 4>ABC?5." ~~~

I check privacy options  for camera, reinstall drivers. Nothing helps

Alexander-Shamin, 2019-07-11:
Update  Firmware Version to 5_11_6_250. Check some drivers:

- Intel RealSense D400 Series UWP Driver v6.1.160.14
- Intel RealSense D400 Series UWP Driver v5.1.160.24
- DMFT_Version 5.1.160.37
Nothing helps

Pristovani, 2019-07-15:
Same in here, I also have similar problems

RealSense Depth Camera D435i
Windows 10 - 64x
SDK 2.24

Samvid95, 2019-07-18:
Hey so do you get this error constantly? Because I got this error a few times when the `pipeline` has not been properly closed the last time. But if I restart the code or replug the camera it gets disappeared. 
Hope that helps.

Alexander-Shamin, 2019-08-08:
I understood what was the matter, windows blocking sdk. 
Also, kasperskiy blocking access to camera.

closed 2019-08-08 by Alexander-Shamin