realsenseai/librealsense #4412: GPS vs T265 position data project

peterwatsonnm, 2019-07-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | Jetson tx2  |
| SDK Version                            |                           | 
| Language                            |                          | 
| Segment			|  Robot                        | 

### Issue Description

I am starting working on a project where I set up a Jetson tx2 & developer kit board and a T265 on a rig that I can carry for a quarter mile or so.  The goal is to gather t265 position data and compare it to gps data. I'll be carrying a gps in my pocket while I walk with the rig. 

 I am trying to figure out the best way to compare the data.  I think I'm going to record it to a .bag file from the T265, and then take a 2d model (as a .png screenshot) of it and scale it and overlay it on the gps map.

I have two questions as of now:  Does this sound like a good way to go about the project?  And, is there any data like this already available somewhere on the web?  I have found this video: https://www.youtube.com/watch?v=YZm6bmGDB5U , but I would like to look at something more concrete than what the video publisher presents.

This is my first time with a project like this, so any advice would be great!

Thanks a lot!

MartyG-RealSense, 2019-07-11:
Others can undoubtedly offer better advice on the technicalities of this subject than I can.  At the time of the T265's launch though, an Intel spokesman told the Venturebeat tech website that the T265 is about as accurate as GPS within 100 meters cubed, if that is of any help.

peterwatsonnm, 2019-07-11:
@MartyG-RealSense Thank you for the information! I have gathered as much from reading online.  I guess my goal is to find out how close "about as accurate" is...

radfordi, 2019-07-12:
Hi @peterwatsonnm,

My guess is that walking will not be nearly as accurate as flying (up high).   When the T265 can see a wide area while it moves it will have less drift than when it keeps seeing new things.  In the end it does dead reckoning.   For example, if you fly close to the ground looking down you will get lots of drift, if you fly up high looking down then you should get less.  My guess is that walking will be in the middle of those two.   We see closed loops errors that are on the order of 1% of the path length.

For example, here's some data from a flight we recently did compared to RTK GPS.
[image]

peterwatsonnm, 2019-07-15:
@radfordi thank you for the detailed explanation and data!  On your graph, what do the colors on the lines indicate?

radfordi, 2019-07-15:
The colors represent the magnitude of the difference of the translation error compared to the RTK GPS ground truth at each corresponding time.

peterwatsonnm, 2019-07-15:
@radfordi thank you!

peterwatsonnm, 2019-07-16:
@radfordi May I ask how you did the overlay of the t265 position data to the gps position data?  I am having trouble coming up with a way to do this.  I am assuming you attached the gps module and t265 to the same board?  As of now my plan is to take the t265 data as a png screenshot and overlay it on a gps map.

radfordi, 2019-07-16:
This data was gathered by setting up a local RTK base station to augment GPS that was gathered on the device along with the T265 data.  These two paths were converted to the same coordinates and then aligned (by minimizing some error between the two paths).  It may not be obvious from the image above, but the data is 3D and the error above includes altitude error, so It's not just "overlayed" in a 2D sense.

asaba96, 2019-07-16:
@radfordi when you collected your data, 
1. How high up did you fly? 
2. How do you have your T265 mounted? Was it pointed straight down at the ground or slightly forward? 

Currently, on our platform, we are utilizing the T265 and pointing it 30 degrees from horizontal (aka straight ahead, but slightly downwards). We sometimes have issues where the pose will jump and remain wrong by some offset, so currently we are utilizing [Robot Localization](https://github.com/cra-ros-pkg/robot_localization) and fusing the velocity from the camera instead of the pose, along with an altimeter and IMU, to get full state estimation.

radfordi, 2019-07-16:
I don't know how high this particular flight was, but usually 20-50m.  The T265 is pointed straight down to avoid seeing the body of the device.  @asaba96, you can now, as of #4321, turn off jumping and/or relocalization.   Let is know how one or the other (or both) work for you.

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

@peterwatsonnm - We plan on closing this issue at this time.  radfordi has answered what we can provide at this time.

closed 2019-09-06 by RealSenseCustomerSupport