realsenseai/librealsense #4413: T265 images over-exposed as compared to D435

fiorano10, 2019-07-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400/T265  | 
| Firmware Version                   | 5.11.6.25 | 
| Operating System & Version |   Ubuntu  | 
| Kernel Version (Linux Only)    |  4.18                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.24                           | 
| Language                            |  C/Python                           | 
| Segment			|  Robot                          | 

When running rqt_reconfigure for the T265 to adjust image params, I am unable to disable auto-exposure, and then adjust the exposure level manually. The lowest value for T265 is 200 while for D435 it's 1. Is there a way to set the exposure level lower for the T265 or should we put a ND filter in front?
Left: D435| Right:T265 (Indoor)

[image]

dorodnic, 2019-07-13:
The auto-exposure of T265 was fine-tuned to provide best quality to the internal SLAM algorithm. When not using our SLAM, you can control exposure and gain manually (try with the Viewer, but also from code). Could you elaborate why do you need lower exposure?

fiorano10, 2019-07-15:
@dorodnic When not using your SLAM, I wasn't able to control the exposure using rqt. I need a lower exposure because the surfaces that we operate on are sandy/reflective. Currently we have physical filters on the D435 to cope with that.

dorodnic, 2019-07-17:
@doronhi - please take a look why this option not working well with ROS

doronhi, 2019-08-11:
using realsense-viewer there seems to be a problem turning the auto-exposure off.
When trying to set auto-exposure off I get the following error message: 
`UNKNOWN in rs2_set_option(options:0x7f8868003ff0, option:Enable Auto Exp Failed set manual exposure`
Later, any attempt at modifying the values results in the following error message:
```
UNKNOWN in rs2_set_option(options:0x7f8868003ff0, option:Exposure. value: ...
To control exposure you must set sensor to manual exposure mode prior to ...
```

doronhi, 2019-08-11:
I'm not sure I understand what the question is.
Are you asking how to turn auto exposure off ? Turning off auto exposure in T265 seems possible, by playing around with realsense-viewer, only prior to start streaming. Therefor, to turn it off in ROS you should set it using ros parameters. Add the following lines to your launch file:
```
  <rosparam>
    camera/tracking_module/enable_auto_exposure: false
  </rosparam>
```
If you run rqt_reconfigure now you'll notice the option `enable_auto_exposure` is now off and you can control exposure and gain.

Are you asking why the lower bound for exposure is 200 and not 1?
I haven't got a good answer for that. This is the device's option's range as you can also see by running `rs-enumerate-devices -o`:
```
Options for Tracking Module
 Supported options:                min        max       step  default
    Exposure                       : 200  ... 16000       0     200
    Gain                           : 1    ... 10          0     1
    Enable Auto Exposure           : 0    ... 1           1     1
```

fiorano10, 2019-08-13:
@doronhi The reasoning behind trying to turn off auto-exposure was to test the lowest exposure value. The auto-exposure on the T265 seems to adjust for closer objects, is it possible to set a `roi`? Can we expect a firmware update to lower the min exposure value.

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Hi @fiorano10

There are no planned changes to the exposure settings or configuration.  The unit for exposure is microsec and config from 200 - 16000.

Let us know if there is anything else needed on this.

Thanks

closed 2019-11-11 by dorodnic