realsenseai/librealsense #4414: Video streams freeze on Jetson Xavier DevKit

WasabiFan, 2019-07-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.06.250 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.108-tegra (JetPack 4.2) | 
| Platform                                 | NVIDIA Jetson AGX Xavier Developer Kit  |
| SDK Version                            |  2.24, 2.18, others                          | 
| Language                            |  All (RealSense Viewer included)                         | 
| Segment			|  Robotics | 

### Issue Description

TL;DR: https://gist.github.com/WasabiFan/5b755a3ecf25771189c571491132527d

I've been chasing this issue for a while (#3268, #3906) and have finally gotten some time to try to try to nail it down.

When streaming color and depth frames from the camera on a Jetson Xavier, after initially streaming for some amount of time without issue, the returned frames will stop updating (frame data is precisely the same after `wait_for_frames`). This usually happens within minutes although I've seen it go for hours without dying in some cases.

This is easiest to see via the demo app I included below but also can be reproduced with RealSense Viewer.

**I have detailed system setup information, a program which demonstrates the issue, and debug logs from failed executions** in [this gist](https://gist.github.com/WasabiFan/5b755a3ecf25771189c571491132527d). You can clone it with git or simply download the ZIP.

Notes on my gist contents:
- The `build-info.md` file contains the tag of the library I built from and the cmake command I used.
- The demo app checks for duplicated frame data and logs when it sees it. When a duplicated frame is received (in this case, checking bit-for-bit identical data in the depth frame), it also logs to the RealSense debug log. You can <kbd>Ctrl</kbd>+<kbd>F</kbd> for `Duped depth frame` in the log file I attached to see where this occurs.
- Larger frame sizes seem to cause more frequent drops.
- I've tested this on three different Jetsons with different software versions and three different RealSenses, with 4+ cables.
- This makes RealSense practically unusable for real systems on the Xavier -- if I can't rely on the video stream staying up, it defeats the purpose of the sensor.

I appreciate a dev looking into this!

### Addendum

In an attempt to circumvent this bug, I have previously tried building with `-DFORCE_LIBUVC=true`. The symptoms are slightly different, but the same bug still seems to occur. In this case it is only the depth frames which stop updating, and the freeze persists for almost exactly five seconds before recovering, only to fail again some time later. With the default v4l backend it never recovers and USB bandwidth drops to zero.

RealSenseCustomerSupport, 2019-08-02:
----------------------------------------------

The build info mentions that the issue happens more frequently when using USB3.2 C-C cable connected to the type C port but less frequent when connected to USB3.0 A port on the Xavier. Have you also tried adding a powered USB 3 C-A hub or PCIe-USB3 raiser card to help to improve the USB stability issue on the Jetson?

WasabiFan, 2019-08-02:
Due to space constraints on my robots, I haven't tried a PCIe card. However, the workaround I'm currently using is an _unpowered_ USB-C hub. I can't say why, but since adding that hub between my Xavier and the RealSense, the problem has largely been mitigated. This is only a stopgap as there's clearly a bug causing it which I'm avoiding.

WasabiFan, 2019-08-04:
To expand on this: there is still broader instability with librealsense even when using the hub (e.g. "frame didn't arrived within 5000", "Error resolving requests") but the freeze issue is gone. I suspect a broader problem with the interaction between RealSense's drivers and the Jetson USB stack, but I need the expertise of those who know the internals of the RealSense to help troubleshoot.

My team has 7 robots each with a Jetson Xavier on it and each with a RealSense camera: these problems make it very difficult for us to actually take advantage of the camera and we spend lots of time resetting the device, unplugging and replugging, etc. I'd really appreciate it if this were looked into. If it's a simple error on our end, even better — it would be nice to know what we can do.

RealSenseCustomerSupport, 2019-08-27:
----------------------------------------------

Since D435 is UVC compliant, it is possible to test them with standard v4l2 test applications such as yavta or the following link to narrow down the issue.
https://github.com/IntelRealSense/librealsense/issues/3416

RealSenseCustomerSupport, 2019-08-27:
----------------------------------------------

@WasabiFan, the frame timestamp and/or frame counter could provide a good way to detect the duplicated frames. The SDK provides a tool - data-collect, to collect and save timestamp. frame num and other information to a file.
https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect
https://github.com/IntelRealSense/librealsense/issues/4551#issuecomment-518364347

A clarification regarding about the previous mentioned v4l test application. They were mentioned to help to isolate any doubt that the frame drop/stop is caused in the RealSense driver/SDK.
My apologies for any confusion this may have caused. Thank you.

WasabiFan, 2019-08-30:
What tests are you asking me to perform -- would you like me to run data-collect and provide the output? Or still do something with v4l?

RealSenseCustomerSupport, 2019-08-30:
----------------------------------------------

Could you please check whether there is any duplicate frame (ID/timestamp) by running rs-data-collect via the USB-C hub first?
`rs-data-collect -c ./data_collect.cfg -f ./log.csv -t 60`
data_collect.cfg:

    DEPTH,1280,720,30,Z16,0COLOR,1280,720,30,RGB8,0

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

Please let us know if there is any problem using the data-collect or further findings. Thank you.

WasabiFan, 2019-09-09:
I will test this when I have some time to go into the lab and set up an Xavier.

Please don't close this issue; it is not resolved.

WasabiFan, 2019-09-19:
With the following setup:

```
$ cat data_collect.cfg 
DEPTH,1280,720,30,Z16,0
COLOR,1280,720,30,RGB8,0
$ rs-data-collect -c ./data_collect.cfg -f ./log.csv -t 600
```

And **NOT** using the USB hub (i.e., in the configuration I know this error occurs), I captured the following log (renamed to `.txt` because GitHub disallows `.csv` extensions):

[log.csv.txt](https://github.com/IntelRealSense/librealsense/files/3633556/log.csv.txt)

The notable section looks like this (there are two of them, this one starting at line 4488):

```
Depth,0,4479,1568934966425.372,149615.395
Depth,0,4480,1568934966458.729,149648.064
Depth,0,4481,1568934966492.087,149681.081
Depth,0,4482,1568934966525.445,149713.082
Depth,0,4483,1568934966558.803,149748.515
Depth,0,4484,1568934966592.161,149781.416
Depth,0,4485,1568934966625.519,149814.864
Depth,0,0,1568934971901.250,155090.109
Depth,0,1,1568934971934.623,155119.281
Depth,0,2,1568934971967.997,155152.572
Depth,0,3,1568934972001.370,155186.660
Depth,0,4,1568934972034.744,155219.258
Depth,0,5,1568934972068.118,155252.502
Depth,0,6,1568934972101.491,155286.310
```

You can see that no depth frames arrive for just over five seconds, and then the frame counter resets and streaming continues. The color stream appears unaffected.

What would you like me to try next?

Using librealsense `v2.24.0` built with `-DFORCE_LIBUVC`.

RealSenseCustomerSupport, 2019-09-21:
----------------------------------------------

Hi @WasabiFan, thank you very much for sharing the test result and findings.
From the log.csv, it looks like the depth stream was restarted.
I will check if this can be replicated here to investigate.
Thank you again for spending time for collecting the data.

WasabiFan, 2019-09-21:
Thanks! That sounds good.

Note that I also included a RealSense debug log in my original post which includes custom markers when duplicate frames were recieved. I don't know whether that would be helpful to figure out more details or not.

WasabiFan, 2019-10-07:
@RealSenseCustomerSupport Any update?

RealSenseCustomerSupport, 2019-10-08:
----------------------------------------------

Hi @WasabiFan,
I have not experienced the issue of the depth id being restarted using the following SDK/FW combo yet.
FW_5.11.15 / SDK_2.29.0
FW_5.11.15 / SDK_2.24.0
FW_5.11.6.250 / SDK_2.29.0
FW_5.11.6.250 / SDK_2.24.0
Attachment is one of the results from above.

Both SDK was built with the following flag
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DFORCE_LIBUVC=true`
Running the data-collect app with the configuration file with D435 directly connected Nvidia Xavier ( L4T-32.2.1) using a single type C to C cable
`DEPTH,1280,720,30,Z16,0`
`COLOR,1280,720,30,RGB8,0`

Thank you.

Attachment(s):
frames_data.csv - https://realsensesupport.intel.com/attachments/token/5Qbn2L7EHwEnWxSVmSw3vy0Ky/?name=frames_data.csv

WasabiFan, 2019-10-12:
I'll try to get another repro.

---

I was reading [this document](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf) and noticed an interesting note:

[image]

We **are not** syncing multiple cameras and don't have those ports wired to anything, But is it possible these are the same issue? This description sounds exactly like what we see.

WasabiFan, 2019-10-21:
OK... I have reproduced this **again**, with a freshly-flashed Jetson, with almost default settings and no special setup.

I flashed this Jetson with stock JetPack 4.2 without any modified settings. There are the only commands I have ever run (the entire terminal history):

```
git clone https://github.com/IntelRealSense/librealsense
cd librealsense/
ls
git tag
git checkout v2.29.0
git log
apt-add-repository universe
sudo apt-add-repository universe
sudo apt-get update
apt-get install libssl-dev libusb-1.0-0-dev pkg-config -y
sudo apt-get install libssl-dev libusb-1.0-0-dev pkg-config -y
cat ~/.bash_history
exit
cat ~/.bash_history
sudo apt-get install build-essential cmake cmake-curses-gui -y
apt-get install libgtk-3-dev libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev -y
sudo apt-get install libgtk-3-dev libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev -y
cd librealsense/
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
mkdir build
cd build
cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CUDA=true -DBUILD_PYTHON_BINDINGS=true -DPYTHON_EXECUTABLE=`which python`
nproc
sudo nvpmodel -m 0
nproc
sudo jetson_clocks 
nproc
make -j7
sudo make install
cd ~/
mkdir freeze-test-2
cd freeze-test-2/
vim data_collect.cfg
rs-data-collect -c ./data_collect.cfg -f ./log.csv -t 600
rs-data-collect -c ./data_collect.cfg -f ./log.csv -t 30
rs-data-collect -c ./data_collect.cfg -f ./log.csv -t 120
history --help
history -w ~/hist.txt
```

I built without any particular modifications to the CMake configuration; as stated in the original issue, I found that when building with stock settings, the depth stream would outright stop and never return, while building with libuvc it would recover after five seconds. This is the former case; my previous tests were with the latter. Both show generally the same issue.

[log1.txt](https://github.com/IntelRealSense/librealsense/files/3753321/log1.txt)

You can see that the depth stream stops before the test does.

Can you reproduce it from these instructions?

WasabiFan, 2019-10-25:
@RealSenseCustomerSupport is this enough to try to reproduce the issue? Is there anything else you need?

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

@WasabiFan, thank you and the steps you provides are quite sufficient for me to test; however, I have not yet seen the freeze issue so far.
Please see the attached results of 30/120/600 seconds of frames using rs-data-collect, depth+rgb, both running at 720p, 30fps.
The D435 is connected directly to Xavier (JetPack4.2) via USB3 type C to Type C cable (3 feet).
I will run additional tests later and update here if anything different.
Thank you.

Attachment(s):
log_30.csv - https://realsensesupport.intel.com/attachments/token/IeyviJiUEsnmGUyWMDK8JvO6D/?name=log_30.csv
log_120.csv - https://realsensesupport.intel.com/attachments/token/EJI1Zk1f5K4d5MRSy2p0pUjbz/?name=log_120.csv
log_600s.csv - https://realsensesupport.intel.com/attachments/token/VthcbVwYO8nIOsRpIp0SAVJiw/?name=log_600s.csv

WasabiFan, 2019-11-03:
@RealSense-Customer-Engineering Did you follow exactly the steps I provided? To match what I did, re-flash the Jetson, then run the shown commands.

I am unable to see the files you linked to (it says "unauthorized" despite being logged in); can you use GitHub file uploads or another normal file sharing service?

In this particular test I used a Type A to Type C cable; it would also be worth testing with precisely that style of cable.

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

Just curious, is this problem still persist on the latest build (2.32.1?) of the RealSense SDK?

WasabiFan, 2020-02-11:
We updated to 2.31 two weeks ago and haven't seen the issue yet but also haven't done extensive testing. I'll post back here when we've had some more time with it under load.

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

Thank you so much for the update.

RealSenseCustomerSupport, 2020-03-18:
----------------------------------------------

@WasabiFan, just wondering if there's any update after the tests?

WasabiFan, 2020-03-23:
I haven't seen this issue occur recently so I'm going to close this. If it arises again I'll re-open the issue.

closed 2020-03-23 by WasabiFan