realsenseai/librealsense #4416: Update firmware of D415

itachi134999, 2019-07-12:
I have a requirement to check the firmware and if it is not recommended update the firmware to recommended firmware.

How to update the firmware from the python code. 

if firmware_update_require:
        //update the firmware from code

MartyG-RealSense, 2019-07-12:
I do not have completed Python code for such a checking function - one of the Intel team on this forum may be able to assist with that.  Through investigating how the SDK software checks the firmware version and suggests the recommended version, I did find a couple of function names that may be useful for your investigation.

RS2_CAMERA_INFO_FIRMWARE_VERSION]
RS2_CAMERA_INFO_RECOMMENDED_FIRMWARE_VERSION

https://github.com/IntelRealSense/librealsense/blob/development/include/librealsense2/h/rs_sensor.h#L25

More info here about firmware checking:

https://github.com/IntelRealSense/librealsense/issues/3818#issuecomment-487403531

Please also note that a new cross-platform firmware update system was introduced in SDK version 2.24.0.  You can read about it in that version's release notes, under the 'New features & improvements' heading.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2240

The standalone **rs-fw-update** tool referenced by the release notes can be found here:

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

itachi134999, 2019-07-12:
@MartyG-RealSense, i can check what is the firmware version of the camera but how can i download and update the firmware from my python code if the version does not match. 
I am using pyrealsense2, does that module provides any inbuilt tool to upgrade or downgrade the firmware ?

MartyG-RealSense, 2019-07-12:
I do not have knowledge of how to script it in Python.  Traditionally (up until the cross-platform firmware updater), firmware update was handled by the DFU tool.  I have not heard of a Python-based method of updating the firmware (again, Intel team members can offer advice about this).

I think that in the cross-platform update system, the firmware updating process may be handled by a script called **fw-update-helper.**

https://github.com/IntelRealSense/librealsense/blob/master/common/fw-update-helper.cpp

I apologize that I cannot be more helpful in this case.

radfordi, 2019-07-12:
Firmware update is now build into `librealsense`.  There is a new command line program `fw-update` that updates the firwmare.  You can take a look at the source to see how to call this yourself.  The summary is to check to see if the `rs2::device` is an `rs2::update_device` and then call `update` with firmware image.  `update_device` is support in python.

radfordi, 2019-07-12:
Here's a quick schematic of how to update a D435 or D435i in python:
```python
import pyrealsense2 as rs
ctx = rs.context()
dev = ctx.query_devices()[0]
ud = dev.as_update_device()
ud.update(open("common/fw/D4XX_FW_Image-5.11.6.250.bin", "rb").read())
```

dorodnic, 2019-07-13:
It's a bit more complicated than that, but generally, yes, `update_device` and `updatable` are the two relevant interfaces. 
First, you need to cast D400 device to updatable and call `enter_update_mode` if I recall correctly. 
Next you need to wait until new device appears and case it to update device.
Then you can call update, and everything should be fine. 
This is however somewhat delicate flow, that I don't encourage integrating directly into your application. Using provided tools (rs-fw-update or the RealSense Viewer) is safer alternative.

itachi134999, 2019-07-15:
@radfordi  I am calling this function:
 upgrade_fw = dev.as_update_device()
 upgrade_fw.update(open("./Intel/Signed_Image_UVC_5_10_3_0.bin", "rb").read())

but i am getting `RuntimeError: null pointer passed for argument "device"`

radfordi, 2019-07-15:
Thanks @dorodnic, I had missed the `updatable` bit.  

Here's an updated "schematic" update process:
```python
import pyrealsense2 as rs
ctx = rs.context()

import threading

dev = ctx.query_devices()[0]
dev.as_updatable().enter_update_state()

import time
time.sleep(1) # using ctx.set_devices_changed_callback(), as in rs-fw-update.cpp, would be better

dev = ctx.query_devices()[0]
dev.as_update_device().update(open("common/fw/D4XX_FW_Image-5.11.6.250.bin", "rb").read())
```

lramati, 2019-08-20:
It looks like the two new classes of `rs2::update_device` and `rs2::updatable` were not properly registered in the python bindings. This may be the cause of some problems. I will fix this for inclusion in a future release.

lramati, 2019-09-02:
This was addressed in PR #4758

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

@itachi134999  Could you please try the latest v2.28 to see if it meets your need? Thanks!

RealSenseCustomerSupport, 2019-09-16:
----------------------------------------------

@itachi13499 Any other questions about this? Looking forward to your update. Thanks!

closed 2019-09-29 by RealSenseCustomerSupport

raghav1906, 2020-06-03:
import pyrealsense2 as rs
ctx = rs.context()

dev = ctx.query_devices()[0]
dev.as_updatable().enter_update_state()

import time
time.sleep(1)

dev = ctx.query_devices()[0]
dev.as_update_device().update(open("Signed_Image_UVC_5_11_6_250.bin", "rb").read())

I used this code to update the firmware but when i executed it, the camera got disconnected and I got an error message saying:
Runtime Error: access denied for 8086:adb uid: 2-3-5
Can someone help me fix this??

avizipi, 2023-02-26:
just for future reference the python code suggested here will not work.
better (and working) python code for firmware update:
`
fw_file = "Signed_Image_UVC_5_12_6_0.bin"
ctx = rs.context()
dev = ctx.query_devices()[0]
with open(fw_file, "rb") as fw:
    dev.as_update_device().update(list(fw.read()))

`