realsenseai/librealsense #4417: Intel RealSense Viewer parameters' documentation

AntoineWefit, 2019-07-12:
### There is a lack of documentation about the different parameters available in the Viewer
I guess this has been notified several times already, but since the SDK has been released for a couple years now it really is a shame that no description is provided for the parameters in the Viewer/the configuration (.json) files.

There **IS** some explanations available here and there, but it really is just to introduce the features to the user. As a "kind of mid-skilled user", it is quite frustrating not to be able to exploit the parameters at there full potential... I'm guessing that we potentially could improve quite a bit our setup if we actually knew what is the behavior of each available parameters. I feel that a ton of them are still obscure to me (and probably other people) and thus simply not used.

**Has the release of such a documentation been planned ?**

Please let me know if missed something that could be useful !

MartyG-RealSense, 2019-07-12:
The 'Advanced Mode' features were not fully documented by Intel because the complex ways in which the interact with each other meant that Intel chose to control them with machine learning instead.  The user is free to do trial and error experimentation with the settings to see how changes affect a stream.  It is not planned to add further documentation for the Advanced Mode functions though.

There is a PDF user guide for the Viewer that collects together information about some of the Viewer's settings into a single document.

https://www.intel.com/content/www/us/en/support/articles/000028593/emerging-technologies/intel-realsense-technology.html

AntoineWefit, 2019-07-12:
Thank you for the fast reply @MartyG-RealSense ! I already downloaded this document, that's the one I was referring to.
I understand that the complexity of the parameters makes it difficult to describe and explain everything, and I don't expressly call for it :)

However, I was hoping for a slight complement of information; like the full name of all the parameters and not just some abbreviation. It would be a nice starting point, and I could then do my own research on the topic !

MartyG-RealSense, 2019-07-12:
I did some extensive research for better plain-English explanations for the parameters but did not have any luck.   The best I could find was the list of Advanced Mode functions in the documentation.

https://unanancyowen.github.io/librealsense2_apireference/classrs400_1_1advanced__mode.html

AntoineWefit, 2019-07-12:
Yes I am aware of this documentation ! I also think that this is the official link : 
https://intelrealsense.github.io/librealsense/doxygen/classrs400_1_1advanced__mode.html

Despite having more explanations, I guess that we at least have a proper list of the parameters...
Thank you for your help though @MartyG-RealSense.

I'll leave this issue open a few days in case of someone has further information to provide.

dorodnic, 2019-07-13:
Hi @AntoineWefit 
Presets are designed to abstract this complexity away from the user and there are some additional ones [here](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets)
When a parameter can be controlled in a useful way, the exact instructions are usually captured in a [white-paper](https://dev.intelrealsense.com/docs/whitepapers).
We could add some extra documentation on couple additional advanced mode parameters, let me check internally with my colleagues.

AntoineWefit, 2019-08-08:
Any update on this @dorodnic ?

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Hi @AntoineWefit
There is a White paper - Tuning depth cameras for best performance - talking about the parameter tuning.
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-play-with-depth-setting

Please let us know if you have further questions.

If we don't hear from you in 7 days, this issue will be closed.

closed 2020-03-10 by RealSenseCustomerSupport