realsenseai/librealsense #4419: Background removal by color

a17sans, 2019-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.11.06 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| Language                            | python | 

Hello, 

I would like to do some background removal but not by depth but by color. 
I mean the example here is an example of background removal for all detected object wich depth greater than 1 meter. I would like to place an object on a black or white background and remove everything of this background. 

Is there a rather simple way to do it ? 

Thank you so much

a17sans, 2019-07-12:
To understand a bit better, I end up with a ply file looking like this : 

[image]

And I would like if possible to do some background removal to get only the banana. 
Thanks !

dorodnic, 2019-07-13:
Seems like a task for OpenCV
There is nothing built in the RealSense SDK for that

a17sans, 2019-07-16:
Or maybe by setting a Region Of Interest (ROI), could it be possible to remove everything that is outside this ROI ? Didn't find anything on not saving certain points on pointcloud but it could be possible

RealSenseCustomerSupport, 2019-08-09:
----------------------------------------------

As dorodnic mentioned, OpenCV could better help in your case.
We will close this case if no further questions.  Thank you.

a17sans, 2019-08-10:
I managed to do it with open 3D with their background cropping : 
www.open3d.org/docs/release/python_api/open3d.geometry.crop_point_cloud.html#open3d.geometry.crop_point_cloud
I will post my code in a moment if somebody has the same issue 
Thanks so much !

closed 2019-08-10 by a17sans