realsenseai/librealsense #4420: Some problems occured when I install the SDK from source

SiuKeungm, 2019-07-12:
Compilation errors are as follows：
**/home/uav1/librealsense/common/fw-update-helper.cpp:15:37: fatal error: common/fw/D4XX_FW_Image.h: not found**
And I can't find the D4XX_FW_Image.h file in the common folde.

radfordi, 2019-07-12:
Firmware update is built in now, so you'll need to run `cmake .` at least once to let it download the firmware.

dorodnic, 2019-07-13:
Also, please make sure there are no CMake warnings related to internet connection, and proxy settings are properly configured. We need to conditionally disable this code when building without internet, but if you want full SDK functionality you need to make sure CMake can download the relevant binaries.

SiuKeungm, 2019-07-16:
@radfordi @dorodnic Solved it! Thanks!

closed 2019-07-22 by dorodnic