realsenseai/librealsense #4424: How to get T265 working with unity in os x?

ArvindSanjeev, 2019-07-13:
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   MacOS  | 
| SDK Version                            |  2.24  | 
| Language                            |  unity    | 
| Segment			|  AR    | 

I was able to successfully run the sample scenes for the D435i in unity on the mac after installing the realsense sdk2.0 (2.24 using brew install) and the wrapper for unity from here: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity . 
The realsense-viewer shows both the D435i and T265 data when opened, however only the T265 slam demo does not work in Unity. It gives me the error:

ExternalException: rs2_pipeline_wait_for_frames(pipe:0x600000163d10)
Rethrow as Exception: Frame didn't arrived within 5000

Is the T265 supported in unity for Mac?

[image]

sserrano44, 2019-07-28:
@ArvindSanjeev could you provide the steps to compile the wrapper on Mac?

ArvindSanjeev, 2019-07-29:
I installed librealsense through brew: `brew install librealsense`
And then downloaded the wrapper from here: https://github.com/IntelRealSense/librealsense/releases/download/v2.24.0/Intel.RealSense.unitypackage
And added it to unity.

RealSenseCustomerSupport, 2019-08-22:
----------------------------------------------

MacOS support for T265 is not supported at this time.

closed 2019-08-22 by RealSenseCustomerSupport

radfordi, 2019-11-08:
It's taken a while, but if you've had problems with T265 on a Mac previously, then #5213 will likely address them.