realsenseai/librealsense #4426: Accuracy problem for rs2_deproject_pixel_to_point.

irvine48, 2019-07-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  NA | 
| Platform                                 | PC |
| SDK Version                           |  2.19.2.688 | 
| Language                               |  C++ | 
| Segment		                 |  others | 

### Issue Description
Hi guys,

I am using D415 to measure object dimensions (width and length), i had go through most of the examples and issue found here and still could not found any solution to my accuracy problem from function **rs2_deproject_pixel_to_point** , i had test both align or not align depth image for that function, both of them give same result. End up i need to do correction by multiple 5%-10% to the **3d_distance** calculated result to match the real dimensions.

Setup of the D415:
1. Point downward to floor, object on floor (0.91m), object height 25.8cm (**accurate**).
2. Disparity shift 39 to filter out floor depth from depth file.
3. Get rectangle of the object (outermost 4 point) and measure distances.
[image]
```
Model: Brown Conrady
[384,312] = X: 7.42076 Y: 7.03714 Z: 65.3
BOTTOM:14.3036CM
Model: Brown Conrady
[248,308] = X: -6.8615 Y: 6.69017 Z: 66
LEFT:14.8204CM
Model: Brown Conrady
[256,168] = X: -6.0258 Y: -8.10631 Z: 66.1
TOP:13.9669CM
Model: Brown Conrady
[388,172] = X: 7.93472 Y: -7.68326 Z: 66.1
RIGHT:14.7511CM
```
**Real dimensions: 26.8 x 15.8 x 14.9 in cm**
I not config anything except disparity shift (filter backgound).

Question 1:
How is the accurate for a single D415 to measure length (in cm? in mm?), 
Question 2:
Do there affected by how height of the object? Seen the height of the object might affect the accuracy of the length measured like change in different ratio so i multiple differently in percents.
Question 3:
Do newer change in SDK correct upper accuracy problem?

irvine48, 2019-07-16:
am i asking stupid question? or no one can drop me a help? guys?

RealSenseCustomerSupport, 2019-08-05:
----------------------------------------------

Hi @irvine48, thank you for checking out the RealSense examples and the forum.
In addition,our depth tuning whitepaper and calibration user guide also discuss in detail the camera accuracy and ways to improve the depth performance.
https://dev.intelrealsense.com/docs/whitepapers?_ga=2.199017391.1884094518.1565045202-1835842910.1558647001

irvine48, 2019-08-06:
> Hi @irvine48, thank you for checking out the RealSense examples and the forum.
> In addition,our depth tuning whitepaper and calibration user guide also discuss in detail the camera accuracy and ways [...]

Thanks for drop this references, i will go through again to see any clue to make the measurement ( distances between to points). Depth is accurate enough in this point.

RealSenseCustomerSupport, 2019-08-09:
----------------------------------------------

Hi @irvine48, glad to hear that you are happy with the D415 depth performance.
Closing this case now if there is no further question. Thank you.

closed 2019-08-20 by RealSenseCustomerSupport