realsenseai/librealsense #4427: How to get depth values from the depth PNG image?

kcshum, 2019-07-14:
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Win 10 | 
| SDK Version                           | 2.24.0 | 

I have converted a recorded bag file to RGB and depth image(PNG format) using rs-converter. Now I want to get the depth values of each pixel directly from the depth image without using the RealsenseD435 device.

I noticed that there is a similar issue posted before. For the code provided(https://github.com/IntelRealSense/librealsense/issues/2200#issuecomment-411774418), 'cout << frame.at(i,j)[0]<< frame.at(i,j)[1]<< frame.at(i,j)[2]<< endl;' gets three values from the png file. 

I am wondering what these three values mean for the depth image. Do I need to do some extra work to get the actual depth value? Do all these three values represent a single pixel?

Looking for your reply.

MartyG-RealSense, 2019-07-14:
If you wanted to get depth data from a saved file, I would be inclined to use the rs-convert tool to convert the bag to .raw format instead of .png.  With PNG, depth data will be be lost when the PNG is created, whilst a .raw file preserves the depth information.

https://github.com/IntelRealSense/librealsense/issues/3640

The link below provides some code for getting depth information from a .raw file.

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331

dorodnic, 2019-07-14:
As discussed here several times before, there is no effective way to get depth back from PNG representation.

kcshum, 2019-07-14:
Thank you for your reply. It solved my problem.

closed 2019-07-14 by kcshum

lichenyue1996, 2019-11-20:
> Thank you for your reply. It solved my problem.

hello, I want to do the same things like you. I want to use realsense to save pictures like TUM dataset.
Could you tell me how to save depth images?