realsenseai/librealsense #4429: Run T265 on RacecarJ

ArtlyStyles, 2019-07-15:
Hi, I have purchased Intel RealSense T265 and would like to install it on my robocar, which has Nvidia Jetson TX2 (https://www.jetsonhacks.com/2018/02/05/software-install). The OS on the TX2 currently is L4T 28.1. Has anyone successfully install T265 on Jetson TX2 with L4T 28.1?

MartyG-RealSense, 2019-07-15:
There was a recent case of a T265 user who installed it on a Jetson TX2 and L4T R28.2.1.  They initially had detection problems, but the T265 worked fine for them once they built with TM2 in CMake (BUILD_WITH_TM2 is ON).

RealSenseCustomerSupport, 2019-08-06:
----------------------------------------------

Hi @ArtlyStyles,

Is the issue clarified and resolved with the correct build option?

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Hi @ArtlyStyles,

I will close this if no additional support required for this ticket.

ArtlyStyles, 2019-08-13:
yes, I was able to install T265 onJetson TX2 with L2T R28. 1

closed 2019-08-15 by RealSenseCustomerSupport